react-native-camera/react-native-camera #1833: RNCamera video quality option Error: Error retrieving camcorder profile params

AsocialCoder, 2018-10-04:
I am using RNCamera and If i use quality option return to me this error;

> Possible Unhandled Promise Rejection (id: 0):
> Error: Error retrieving camcorder profile params
> Error: Error retrieving camcorder profile params
> 

```
 const options = {quality:RNCamera.Constants.VideoQuality["480p"]}
        const { uri, codec = "mp4"  } = await this.camera.recordAsync(options);
        this.setState({ recording: false, processing: true });
        const type = `video/${codec}`;
        const data = new FormData();
        data.append("video", {
          name: "mobile-video-upload",
          type,
          uri
        });
        console.log(type);
}
```

AsocialCoder, 2018-10-04:
what kind of information do you want?

AsocialCoder, 2018-10-04:
> I am using RNCamera and If i use quality option return to me this error;
>
>
> ```
>  const options = {quality:RNCamera.Constants.VideoQuality["480p"]}
>         const { uri, codec = "mp4"  } = await [...]

reime005, 2018-10-04:
The more you can provide, the easier to reproduce. Infos about the device mainly

sibelius, 2018-10-04:
fixed here https://github.com/react-native-community/react-native-camera/pull/1835

try master

closed 2018-10-04 by sibelius