react-native-camera/react-native-camera #1834: Unable to scan Compact/Truncated PDF417 barcodes on iOS

D3athSpank, 2018-10-04:
# Bug Report

**Platforms**
iOS

**Versions**
- iOS:
- react-native-camera: 1.3.0
- react-native: 55.4
- react: 16.3.1

**Description/Current Behaviour**

Unable to scan a Compact/Truncated PDF417 barcode

**Expected Behaviour**
Scanning a Compact/Truncated PDF417 should give a barcode back

**Does it work with Expo Camera?**
No, this is the reason we ejected out of Expo, to test the react-native-camera purely

closed 2019-06-08 by stale[bot]

dondragon2, 2020-10-28:
@D3athSpank were you able to resolve this issue. This is one year later and I am experiencing this issue. Any help is welcomed

D3athSpank, 2020-10-28:
> @D3athSpank were you able to resolve this issue. This is 2 year later and I am experiencing this issue. Any help is welcomed

No, sorry Im also out of that project using this