react-native-camera/react-native-camera #1838: Option to Not Automatically Resume Preview After Capture

FullstackJack, 2018-10-08:
# Feature-Request

**Describe the Feature**
Some camera apps immediately launch a new screen to preview and edit the image before upload (i.e. Snapchat, Instagram). Unfortunately, this package automatically resumes camera preview before returning from onAsyncPictureTaken (which is usually used to navigate). The result is that the user sees an intermittent camera preview before the navigation action. This behavior is confusing and should be optional by function options key.

**Possible Implementations**
Wrap the resumePreview call in an if statement.

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

SimonErm, 2018-10-10:
Is this just related to Android?

FullstackJack, 2018-11-05:
@SimonErm The code I referenced is on Android. I haven't an iOS device to test this on that platform.

SimonErm, 2018-11-21:
#1641 Works on latest Release

closed 2018-11-21 by SimonErm