react-native-camera/react-native-camera #1842: Capture Photo is too slow in Android, takes around 3 seconds to capture

harishbbb, 2018-10-10:
Hi,

If any one has solution, please let me know

**Camera config is**

<Camera
      ref={(cam) => {
        this.camera = cam;
      }}
      style={styles.preview}
      captureTarget={Camera.constants.CaptureTarget.temp}
      captureMode={Camera.constants.CaptureMode.still}
      aspect={Camera.constants.Aspect.fill}
      type={Camera.constants.Type.back}
      orientation={Camera.constants.Orientation.portrait}/>

**For Capturing Picture:**

takePicture() {
    const options = {}
    this.camera.capture({metadata: options})
    .then((data) => this.afterCapture(data: data))
    .catch(err => console.error(err));
  }

It is working good for iOS(max 1 second to capture photo)

reime005, 2018-10-10:
See #1836

SimonErm, 2018-10-10:
RCTCamera is deprecated use RNCamera instead.

anjmao, 2019-08-01:
@n1ru4l How can this issue be duplicate of itself?

n1ru4l, 2019-08-01:
@anjmao I referenced the wrong issue. #1836 is correct