react-native-camera/react-native-camera #1844: Error while updating property 'googleVisionBarcodeDetectorEnable'

rafirr, 2018-10-10:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**

RNCamera

**Platforms**

Android

**Versions**

- Android: 6.0.0
- react-native-camera: 1.3, 1.2, master
- react-native: 0.57
- react: 16.5.1

**Steps to Reproduce**

```js
<RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={{
            flex: 1,
            alignItems: 'center',
            flexDirection: 'column',
            justifyContent: 'space-between'
          }}
          type={this.state.type}
          flashMode={this.state.flash}
          autoFocus={this.state.autoFocus}
          zoom={this.state.zoom}
          ratio={this.state.ratio}
          focusDepth={this.state.depth}
          onGoogleVisionBarcodesDetected={
              (e) => {
                  console.warn(e)
              }
          }
        >
...
</RNCamera>

```

**Additionals**

[image]
https://github.com/react-native-community/react-native-camera/pull/1457
Any idea how fix to this bug?

tonybebber, 2018-11-15:
@rafirr - I'm getting the same error. Were you ever able to figure out why this error is occurring?

2f-concept, 2018-11-20:
@rafirr - I'm getting the same error too, I've found a solution to sole this ?

n1ru4l, 2018-11-20:
@rafirr @tonybebber  @2f-concept Could anyone of you please provide a reproduction repository?

2f-concept, 2018-11-20:
@n1ru4l I've uninstalled then reinstalled manually this plugin and never seen this message again.

sibelius, 2018-11-20:
tks

closed 2018-11-20 by sibelius

tonybebber, 2018-11-20:
I'm not sure if this was the proper fix, but I got things working by updating the dependencies section in `/android/app/build.gradle`. When I updated the com.google.android.gms:play-services-vision dependency from 12.0.1 to 15.0.2, the error went away.

Here is what the relevant code in `dependencies` looks like for me, now that things are working as expected.

```
    compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        compile 'com.android.support:exifinterface:25.+'
        compile ('com.google.android.gms:play-services-vision:**15.0.2**') {
            force = true
        }
        compile ('com.google.android.gms:play-services-base:16.0.1') {
            force = true
        }
    }
```

2f-concept, 2018-11-21:
@tonybebber When I apply Your changes to my `/android/app/build.graddle` the error come back...
It seem realy to be an issue around this but I don't know how to set the proper version.
Here is the code that work for me:
`implementation (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        implementation 'com.android.support:exifinterface:26.+'
        implementation ('com.google.android.gms:play-services-vision:12.0.1') {
            force = true
        }
    }`

rafirr, 2018-11-21:
@tonybebber, @2f-concept for me that's working:

```
// android/app/build.gradle

compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        compile 'com.android.support:exifinterface:25.+'
        compile ("com.google.android.gms:play-services-vision:${rootProject.ext.googlePlayServicesVersion}") {
            force = true
        }
}
```

```
// android/build.gradle

ext {
    buildToolsVersion = "27.0.3"
    minSdkVersion = 19
    compileSdkVersion = 27
    targetSdkVersion = 26
    supportLibVersion = "26.1.0"
    googlePlayServicesVersion = "15.0.1"
    firebaseCoreVersion = "16.0.1"
    androidMapsUtilsVersion = "0.5+"
}
```