react-native-camera/react-native-camera #1846: Orientation not working in video

DanthonyPabon, 2018-10-10:
>  "react-native": "0.56.0",
>  "react-native-camera": "^1.3.0",
>   node: v10.11.0

# Question
I'm using this to take a video, but the orientation is not working, the video doesn't take the orientation

```
async startRecording() {
    this.setState({ recording: true });
    const data = await this.camera.recordAsync({
      quality: RNCamera.Constants.VideoQuality["480p"],
      orientation:"landscape"
    });
  }
```

SimonErm, 2018-10-10:
On Android or iOS? The orientation option is just implemented for iOS right now.

DanthonyPabon, 2018-10-10:
Android

SimonErm, 2018-10-10:
Like i said, orientation option is just available for iOS right now. Feel free to make a PR.

closed 2018-10-10 by SimonErm