react-native-camera/react-native-camera #1848: [iOS] App crashes after recording video through front camera

BaderSerhan, 2018-10-11:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

Issue started to appear after running `react-native-update`.
In my `package.json` file, I have `"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git"`, so I am connected to master.

**Related Modules**

RNCamera

**Platforms**

iOS

**Versions**

- react-native-camera: 1.3.0
- react-native: 0.56.1
- react: 16.3.1

**Description/Current Behaviour**

Front camera video is crashing after the `stopRecording` function is called. However, back camera video is working perfectly okay!
When I debug the issue, the `await this.camera.stopRecording(options)` is never resolved; I am not even getting an error in the `catch` block, or in the console... Nothing at all to help me :/

**Expected Behaviour**
Recording video through the front camera should work exactly like through back camera.

**Steps to Reproduce**
If it might help, here is my `recordAsync` function:

    recordAsync = async function () {
        if (this.camera) {
            const options = {
                mirrorVideo: this.type === RNCamera.Constants.Type.front,
                quality: RNCamera.Constants.VideoQuality["720p"],
                // mute:false,
                audio: true,
                captureAudio: true,
                maxDuration: 30,
                base64: true,
                forceUpOrientation: true,
                fixOrientation: true,
            };
            console.log('waiting for data')
            const data = await this.camera.recordAsync(options)
                .catch((err) => console.log("promise error", err))
            console.log("video data", data)
            return data
        } else {
            alert("no video data")
        }
    };

Petka17, 2018-10-15:
We have the same issue. Right now we upgraded Xcode to 10th version, RN to 0.57.2 and react-native-camera to 1.3.0. On iOS 12 everything is working fine but on iOS 11 app crash at the record start.

PS: no problem on android

BaderSerhan, 2018-10-16:
**[UPDATE]**

@SimonErm Thank you for your help, I have set mirrorVideo to false and the app is no longer crashing!
But we need a fix for this soon cause we would like our front camera videos to be inverted.

Thank you for your tremendous work!

SimonErm, 2018-10-17:
The problem is that there is no rejection if mirrorVideo fails at https://github.com/react-native-community/react-native-camera/blob/8de827e6c2a027b30668c108b147fb02affb8a35/ios/RN/RNCamera.m#L990 . The other question to ask is why it fails.

Petka17, 2018-10-18:
We passed `quality` option to recordAsync and the application is no longer crashing on iOS 11.

```
this.camera.recordAsync({
    quality: RNCamera.Constants.VideoQuality['720p']
});
```

package.json
```
...
"react-native": "0.57.2",
"react-native-camera": "^1.3.0",
...
```

Xcode Version 10.0 (10A255)

SimonErm, 2018-11-20:
@Petka17  Cann you try if this is fixed without workaround in latest release? #1908

jtwarren, 2018-11-30:
I'm still seeing this crash on 1.4.3 with both quality and mirrorVideo defined:
```
{
  ...,
  quality: RNCamera.Constants.VideoQuality["720p"],
  mirrorVideo: true 
}

BaderSerhan, 2018-12-01:
@jtwarren you should set `mirrorVideo: false` for it to work

n1ru4l, 2018-12-02:
@BaderSerhan Is the `mirrorVideo` option broken?

jtwarren, 2018-12-02:
@BaderSerhan I specifically want `mirrorVideo: true`, and yes @n1ru4l it is crashing for me on 1.4.3.

BaderSerhan, 2018-12-04:
@n1ru4l yes, the crash is occurring because of `mirrorVideo: true` option.
@jtwarren I want it too, but I'd rather have an inverted video in my app for now rather than a crash.

closed 2019-06-08 by stale[bot]

MrPluto, 2019-08-09:
seems work well on my iOS 12.4, I'm using version 2.9.0

MrPluto, 2019-08-09:
2.9 works fine on my iOS 12.4