react-native-camera/react-native-camera #1856: Problems with implementing TextDetector for iOS

howardwkim, 2018-10-14:
Upon entering `RNcamera` screen, I get a `FaceDetector not integrated, stub used!` warning. Then `onTextRecognized` is immediately called, returning  `{type: "TextBlock", target: 75, textBlocks: Array(0)` without actually detecting any text. They seem related.

* How do I resolve the issues related to the `FaceDetector not integrated, stub used!` warning?
* Is there an example implementation of TextDetector for iOS?

### Steps to Reproduce
1. `"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git"`
2. `rm -rf node_modules && npm install`
3. Add the following to my pod file, then `pod upgrade`
```
  pod 'GoogleMobileVision/TextDetector'
  pod 'react-native-camera', :path => '../node_modules/react-native-camera', subspecs: [
    'TextDetector'
  ]
```
4. Start app and go to screen with `RNCamera` implementation.

### Expected Behavior
Upon detecting text, returns `TextBlock` object  (hopefully with detected text)

### Actual Behavior
Upon entering `RNcamera` screen, I get a `FaceDetector not integrated, stub used!` warning. Then `onTextRecognized` is immediately called, returning  `{type: "TextBlock", target: 75, textBlocks: Array(0)` without actually detecting any text.

### Environment
`"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git",`
`"react-native": "^0.56.0"`
`"react": "^16.4.1"`

dov11, 2018-10-18:
Hi @howardwkim, #1863 should remove the warning, while you can check out a crude example of Text detector implementation here: https://github.com/dov11/rncamerademo. Hope that helps.

howardwkim, 2018-10-21:
@dov11 Thank you! I got iOS text detection working with your change in #1863 , and your example was helpful as well.

closed 2018-10-21 by howardwkim

ravik2015, 2021-08-12:
Hello, In my case the onTextRecognized function never gets called in IOS device, the same thing is working on android.
My app details are:
 react-native 0.64.2
 react-native-camera 4.0.1