react-native-camera/react-native-camera #1859: Code smells and compile warnings in iOS native code

thedjinn, 2018-10-16:
# Bug Report
**To Do First**
- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Related Modules**
RNCamera and probably others.

**Platforms**
iOS

**Versions**
RN 0.57, react-native-camera 1.3.0, Xcode 10.0

**Expected Behaviour**

The RNCamera project produces lots of compile time warnings (33 in my case). After investigating I noticed several code smells that should be addressed. 

1. When compiling for simulator there are several early returns surrounded in `#if` blocks. This causes "Code will never be executed" warnings. The remainder of the function should be wrapped in an `#else` block.

2. Unused variables outside of GoogleMobileVision `__has_include` blocks: there are `Class`-type local variables introduced in several methods regarding face detection that are only used inside an `#if` block. The declarations should be moved inside the `#if` block to prevent unused variable warnings.

3. `RNCamera.h` defines methods that are only implemented inside `#if` blocks. The methods should always be provided but their implementations should be wrapped in `#if` blocks instead so code can never crash on executing a non-available method.

4. There are implicit conversion warnings. Probably because of API changes in React Native. They can be silenced by casting to the proper type because precision loss is never an issue as far as I could see.

Additionally, the static analyzer found an API misuse error that might hint at a possible bug.

n1ru4l, 2018-10-26:
Hey @thedjinn, it seems like you are quite familiar with objective c. Would you like to remove these code smells in a pull request? Also do you know any tools that could prevent those issues, e.g. by using a code analyzer at commit time? Ideally the tool is a cli and can be run on circle ci (without requiring a mac osx environment)

SimonErm, 2018-11-14:
Maybe we can use https://sonarcloud.io/about/sq . It's free for open source projects.

closed 2019-06-08 by stale[bot]