react-native-camera/react-native-camera #1861: iOS Facial Detection Currently Broken (iOS 12)

jbwhitcombe, 2018-10-17:
# Bug Report
**To Do First**
- Tried latest release
- Tried master
- There are existing reports however they are all closed

**Related Module**
FaceDetector

**Platform**
iOS

**Description/Current Behaviour**
Currently crashing with iOS 12 with an EXC_BAD_ACCESS when onFacesDetected function is called.

**Expected Behaviour**
Should not cause a crash. This bug has rendered the library unusable for facial recognition on iOS.

**Steps to Reproduce**
On an iPad 10.5 model, instantiating an RNCamera view and supplying a function for OnFacesDetected causes a crash immediately. 

[image] with the following error appearing in Xcode.

jgfidelis, 2018-10-18:
There was an issue where people identified what line code broke the facial recognition. Do you know enough objective C to try and fix it?

jbwhitcombe, 2018-10-18:
Perhaps! My Objective-C > Swift, although any direction on what line is causing this would help!

n1ru4l, 2018-10-18:
duplicate of https://github.com/react-native-community/react-native-camera/issues/1613

closed 2018-10-18 by n1ru4l

imspikey, 2018-10-18:
yes but  #1613  is closed without solution?!!!

jbwhitcombe, 2018-10-22:
@n1ru4l Yep. This resolves this issue for sure, thanks heaps! Interestingly, I had to reinstall everything via Pods as manual installation of the GMV didn't work. Perhaps it's worth adding this as part of the installation process for the latest version and with iOS 12 now?