react-native-camera/react-native-camera #1862: Rendering in response to state changes causes camera to go black

petlys, 2018-10-17:
### Which implementation are you using

RNCamera

### Steps to reproduce

1. Open camera
2. Scan barcode which updates state
3. Conditionally render a component inside the camera view in response to the updated state

### Expected behaviour

Component should render as expected when state changes

### Actual behaviour

Only an issue on Android.

When `onBarCodeRead` is fired, I update a state which causes a component to render.
When that component renders, the camera screen goes black. 

```javascript
const styles = StyleSheet.create({
  container: {
    display: 'flex',
    flex: 1,
    justifyContent: 'center',
    flexDirection: 'column',
    alignItems: 'center'
  }
})

...

<RNCamera
  onBarCodeRead={() => this.setState({ success: true })}
  permissionDialogTitle="Permission to use camera"
  permissionDialogMessage="We need your permission to use your camera."
  style={styles.container}
>
  {this.state.isSuccess && <Text>Great success!</Text>}
</RNCamera>
```

### Environment

- React Native version: 0.48.1
- React Native platform + platform version: Android 7.0
- react-native-camera version: 1.3.0

Samsinite, 2018-10-18:
Getting this same error, on React 16.3.1, RN 0.55, android, and react-native-camera 1.3.0

Samsinite, 2018-10-18:
For some reason it seems to work if I have `flex: 0` styles on RNCamera initially. If I change it via the inspector after to `flex: 1` then it continues to work. Not sure what is going on...

petlys, 2018-10-18:
Thanks for sharing! I successfully worked around the issue by having `RNCamera` as its own component, and using absolute positioning to overlay it:

```javascript
<View style={{ flex: 1 }}
  <RnCamera style={{ flex: 1 }} />
  {this.state.isSuccess && <Text style={{ position: absolute; top: 0; left: 0; right: 0; bottom: 0 }}>Great success!</Text>}
</View>
```

`z-index` doesn't seem to work properly on Android, so you have to utilise the order of components to decide what goes on top of what.

Samsinite, 2018-10-18:
Thanks! I might try to help get this fixed this weekend and prefer to keep using facc for now as the app is paying attention to the status. Is there a way to grab the camera status if not using facc?

miracle2k, 2018-10-26:
I ran into the same problem, and struggled for quite some time with permissions (https://github.com/react-native-community/react-native-camera/issues/1752), which turned out to be a dead end.

I was stepping through the code with a debugger to try to figure it out. What seems to happen is:

- When the state update happens,  react-native determines that it needs to run a `UIOperation` to delete a particular node. 
- `NativeViewHierarchyManager.java:manageChildren` is invoked in such a way that the line `viewManager.removeViewAt(viewToManage, indexToRemove)` is called with `viewToManage` being `RNCameraView` and  `indexToRemove` being 0.
- This makes sense to be from a React perspective - I do want to remove my first (and  only child view).
- However, on the Android side, the first child is actually the TextureView of the camera, which is now being destroyed.

miracle2k, 2018-10-26:
Note that for me this happens even if I do not use any custom state  updates myself; I am only changing the child render based on the `status` property being passed down to me, just like in the example.

At this point I am pretty sure this is a timing issue. The simpler I make my `PendingView` and the app startup code, the less  likely it is to appear; but I am seeing at least on some occasions, with the same code, this bug appearing as well as not.

My guess is that the logic I described above is the bug (React is not aware of the implicit child added by the camera view). This bug is trigged reliably during a state update, and it can be triggered with the default example code (the function-as-a-child component updating itself) if the timing is unfortunate.

miracle2k, 2018-10-27:
Just run across this comment which may be react-native-maps facing a similar problem: https://github.com/react-community/react-native-maps/blob/master/lib/ios/AirMaps/AIRMap.m#L53

n1ru4l, 2018-12-19:
@miracle2k You are referring to Android Code, but the actual report is related to android.

@petlys @miracle2k Could check if your issue is resolved when using this pull request: https://github.com/react-native-community/react-native-camera/pull/1995

miracle2k, 2018-12-23:
@n1ru4l The link to the `react-native-maps` repo indeed goes to iOS code, but the investigation I described above I did in the Android debugger, and I ended up feeling fairly certain (like 80% certain) that this is the underlying issue here.

`react-native-maps` apparently doesn't have this problem on the Android side, because there, `<MapView>` just creates a single native view. On the `iOS` side, the comment suggests that the native map view from Apple does have automatic subviews. 

`react-native-camera` on Android seems to run into the same issue: There, the native camera view has a subview (TextureView).

I am fairly certain that the pull request #1995 will fix it, because it avoids the issue described.

sibelius, 2019-01-08:
fixed here https://github.com/react-native-community/react-native-camera/releases/tag/v1.7.0

closed 2019-01-08 by sibelius