react-native-camera/react-native-camera #1864: [Video Record] has no sound

fokoz, 2018-10-19:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
iOS

**Versions**

- Android:
- iOS: 11
- react-native-camera: 1.3.0
- react-native: 0.56.1
- react: 16.4.1

**Description/Current Behaviour**
Record video with no sound even set `captureAudio=true`, have tested with VLC player in the PC and also with `react-native-video `no sound at all.

**Expected Behaviour**
The sound should be recorded.

**Steps to Reproduce**
1. Record video with `captureAudio=true` and allow the permission.
2. Video has only image with no sound.

fokoz, 2018-10-19:
Here's file example:
[qlZTYpiaQbcdOZPzAHhWYD7nNZRRmDSCFUHFlxbg.mov.zip](https://github.com/react-native-community/react-native-camera/files/2496683/qlZTYpiaQbcdOZPzAHhWYD7nNZRRmDSCFUHFlxbg.mov.zip)

ryekerjh, 2018-11-16:
@fokoz Did you ever find a solution for this?

fokoz, 2018-11-18:
@ryekerjh don’t set mute in record options. Even if you set muted=false it will be muted anyway.

ryekerjh, 2018-11-19:
@fokoz My coworker found that shortly after I tagged you. Thanks so much for responding!

n1ru4l, 2018-11-19:
:tada: This issue has been resolved in version 1.4.2 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.4.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #1932 (2018-11-19): fix(ios): only mute recording if mute property is set to false
Closes #1864
commit 776725ea (2018-11-19): fix(ios): only mute recording if mute property is set to false (#1932)

Closes #1864
closed 2018-11-19 by n1ru4l