react-native-camera/react-native-camera #1866: Android not compiling when using AndroidX libs

markmckim, 2018-10-20:
# Bug Report
**To Do First**
- [X] Did you try latest release?
- [ ] Did you try master?
- [X] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
Android

**Versions**
- Android: compileSDK 28, buildTools 28.0.3
- react-native-camera: 1.3.0
- react-native: 0.56.0
- react: 16.4.1
- react-navigation: 2.9.1

**Description/Current Behaviour**
I'm using AndroidX libs throughout my app (i'm also using Kotlin, but i don't believe that is related), and  when following the steps outlined here: https://www.npmjs.com/package/react-native-camera, the app will not compile. Getting over 50 errors at build time relating to not being able to find various libs:

```error: package android.support.v4.content does not exist```

I've tried nearly every solution i've found online, including adding support libs to build.gradle files etc. but i'm not able to get this to compile. It's quite possible i've missed something, but it may also be an AndroidX issue.

Any help much appreciated! 

**Expected Behaviour**
App should build successfully

**Steps to Reproduce**
See description

chrismowbray92, 2018-10-30:
Getting the same issue! Can this get looked at ASAP?

kassim, 2018-11-21:
I can confirm creating a `/node_modules/react-native-camera/android/gradle.properties` file with
```
android.useAndroidX=false
android.enableJetifier=false
```
fixes the build for me
so this needs to be generated by the https://github.com/react-native-community/react-native-camera library then to fix the problem

soumyamishra89, 2019-01-14:
is there any progress on this issue? With the new androidx migration the app build is stalled.

kassim, 2019-01-14:
@soumyamishra89 if you point to our branch in your **package.json** like so:
```
  "dependencies": {
    "react-native-camera": "side-co/react-native-camera#AndroidX"
```
and `npm install` then it should work for you

I haven't synced it with the latest updates in a while
perhaps you may want to fork your own branch to be able to apply those updates

markmckim, 2019-06-04:
This is still an issue preventing me from migrating my project to AndroidX

sibelius, 2019-06-04:
check this https://github.com/react-native-community/react-native-camera/issues/2296

and https://github.com/react-native-community/react-native-camera/pull/2068#issuecomment-498781811

androidX is shipped on react-native-camera already

markmckim, 2019-06-04:
> check this #2296
> 
> and [#2068 (comment)](https://github.com/react-native-community/react-native-camera/pull/2068#issuecomment-498781811)
> 
> androidX is shipped on react-native-camera already

Fantastic! Looks like I just missed this over the last few days. That's great news. Thanks for the info!!

laurent22, 2019-06-14:
I'm still having this issue even after upgrading to rn-camera 2.10.2:

```
> Task :react-native-camera:compileGeneralDebugJavaWithJavac FAILED
/Users/laurent/src/joplin/ReactNativeClient/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/RNCameraView.java:10: error: package android.support.v4.content does not exist
import android.support.v4.content.ContextCompat;
                                 ^
/Users/laurent/src/joplin/ReactNativeClient/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:27: error: package android.support.annotation does not ex
ist
import android.support.annotation.IntDef;
                                 ^
/Users/laurent/src/joplin/ReactNativeClient/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:28: error: package android.support.annotation does not ex
ist
import android.support.annotation.NonNull;
```

Does something need be done in order to get it to build with AndroidX?

markmckim, 2019-06-14:
Yeah I got the same result. I think they had to roll back those changes due to an issue and are going to release Android X support under a 3.0.0 version instead.

felansu, 2019-06-17:
same problem

t1amat9409, 2019-06-19:
``` npm jetifier && npx jetify``` kinda helped me

7laria, 2019-06-19:
same problem

mike-niemand, 2019-06-19:
Same here. This needs to be reopened.

felansu, 2019-06-20:
> ` npm jetifier && npx jetify` kinda helped me

not good...

doancongbang1991, 2019-06-21:
`npm install --save-dev jetifier (or use yarn, but install it locally in your project, not globally)
npx jetify or npx jetify -w=1 (to specify the number of parallel workers)
npx react-native run-android`

closed 2019-08-27 by stale[bot]