react-native-camera/react-native-camera #1869: Does not recognize barcodes on lighter space (examples in links)

nadzic, 2018-10-22:
**Platforms**
Android, iOS

**Versions**
- Android: 8.1.0
- iOS: 11.2
- react-native-camera: ^1.1.2
- react-native: 0.52.0
- react: 16.2.0

**Description/Current Behaviour**
If you are trying to scan barcode in the light grey color and on little bit lighter space the react native camera does not recognize the barcode. We tried with other barcode scanning apps and they recognize it - so it is problem react native camera.

Examples of barcodes (they are UPCE type):

https://files.slack.com/files-pri/T36AT99BQ-FDJARJ0Q3/image_from_ios.jpg
https://files.slack.com/files-pri/T36AT99BQ-FDJKEMGCV/image_from_ios.jpg
https://files.slack.com/files-pri/T36AT99BQ-FDJDL2XSN/image_from_ios.jpg

But it recognizes in white/black contract, for example this one:

https://barcode.tec-it.com/en/UPCE?data=07811403

**Expected Behaviour**
It should scan barcode successfully.

**Steps to Reproduce**
Just try to scan the barcode with react native camera.

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

n1ru4l, 2018-11-20:
The link to the barcodes that cannot be recognized does not work since they are from a private slack. You can upload images on github by drag'n'dropping them into you comment.

If you want to tackle this issue please feel free to send pull requests!

IssuehuntBot, 2018-12-03:
@issuehuntfest has funded $20.00 to this issue. [See it on IssueHunt](https://issuehunt.io/repos/33218414/issues/1869)

closed 2019-06-08 by stale[bot]