react-native-camera/react-native-camera #1870: Facedetector is not detecting any faces (iOS)

olegwn, 2018-10-23:
# Bug Report
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**
RNCamera
FaceDetector

**Platforms**
iOS

**Versions**
- iOS: 11.x, 12.x (tried on two different devices)
- react-native-camera: master (fc3f7bd)
- react-native: 0.57.3
- react: 16.6

**Description/Current Behaviour**

When I open a frontal camera I get initial callback on onFacesDetected with:
`{type: "face", faces: Array(0), target: 347}`

After that, when I direct camera to face or faces the callback is never called again.

**Expected Behaviour**
Should detect faces and call onFacesDetected callback

**Steps to Reproduce**
Installed Facedetector using CocoaPods with:
` pod 'react-native-camera', subspecs: ['FaceDetector'], path: '../node_modules/react-native-camera'`

My RNCamera settings:

```            type={RNCamera.Constants.Type.front}
            autoFocus
            flashMode={RNCamera.Constants.FlashMode.auto}
            faceDetectionLandmarks={RNCamera.Constants.FaceDetection.Landmarks.all}
            onFacesDetected={this.onFacesDetected}
            onFaceDetectionError={this.onFaceDetectionError}
            faceDetectionMode={RNCamera.Constants.FaceDetection.Mode.accurate}
```

Tried with RNCamera.Constants.FaceDetection.Mode.fast as well and RNCamera.Constants.FaceDetection.Landmarks.none
makes no difference

olegwn, 2018-10-31:
strange thing, but after recompiling project in XCode 10 with legacy build system the face detection now works!

jgcmarins, 2018-10-31:
can we close?

brunotalhate, 2018-12-06:
I've recompiled using legacy build but I'm still facing the `{type: "face", faces: Array(0), target: 347}` object.

Using cocoapods, iPhone 6 with iOS 12, xCode 10.1 and couldn't find any solution yet

UberMC, 2018-12-08:
Is face detection broken on iOS?

brunotalhate, 2018-12-08:
I've been unable to make it work on iOS using both cocoapods and non-cocoa.
Spent 2 days trying a lot of specs without success. 😞

The closest I've achieved was the empty face on mount

Did your tried it lately?

olegwn, 2018-12-09:
I noticed that face detection works on iPhone X device but doesn't work on iPhone 6 device!

brunotalhate, 2018-12-12:
Tried with manual linking and I was able to build without errors using the package version `"react-native-camera": "git+https://github.com/react-native-community/react-native-camera.git#7d172af864e963fde3d732fbfd2b8d124ab2b582"` and `Legacy Build System`

sibelius, 2018-12-12:
@brunotalhate can we do a git bisect to check where the bug was introduced?

syadykin, 2018-12-13:
Commenting out https://github.com/syadykin/react-native-camera/blob/master/ios/RN/RNCamera.m#L622 reverts face detection back, not sure what it is for though.

n1ru4l, 2018-12-13:
@syadykin It is for setting the `defaultVideoQuality` property on the `RNCamera` instance in order to prevent flickering. The flickering cannot be avoided when using `recordAsync`

telmen, 2019-01-03:
Any update on this? using cocoapods

telmen, 2019-01-03:
Hi @brunotalhate, any luck with it?

tiendn, 2019-01-03:
Hope !

srGritsuk, 2019-01-09:
**It broke right here:**
(Older devices cannot shoot video with photo quality)
Its work fine from iPhone 6s and earlier with:
self.session.sessionPreset = AVCaptureSessionPresetHigh (not AVCaptureSessionPresetPhoto)

https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L623

You can manually set a weak preset, for iPhone 6s works fine preset:
defaultVideoQuality={ RNCamera.Constants.VideoQuality.720p }

https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md#ios-defaultvideoquality

telmen, 2019-01-10:
Yeah, it seems solving the problem.

sibelius, 2019-03-25:
can more people test these 2 PR

https://github.com/react-native-community/react-native-camera/pull/2044
https://github.com/react-native-community/react-native-camera/pull/2036

which one makes more sense?

Thomazella, 2019-03-28:
@sibelius idk, I tried both and neither work. Also text and barcode don't work for me. It's not calling the `onFacesDetected` callback. iPhone 6s. Maybe I did something wrong? should I expect to see some points render on the screen if my face is detected?

Thomazella, 2019-03-28:
also tried commenting out the line https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L623
and tried setting it to `AVCaptureSessionPresetMedium`

sibelius, 2019-03-28:
Have you setup face detection on iOS? Are you using manual link or pods?

Thomazella, 2019-03-28:
Gonna try again, apparently had a setup issue. Using manual link.

n1ru4l, 2019-04-11:
:tada: This issue has been resolved in version 2.3.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v2.3.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

luvnish19, 2020-07-02:
I am using Xcode 11.3.1 and testing on iPhone 7 with OS 13.3.1. Both onFacesDetected & onFaceDetectionError methods are not called. Can someone have any solution.
I will be very thankful.

locliem08, 2020-07-22:
I am using the latest current version: "react-native-camera": "^ 3.32.0". Face recognition doesn't work on the rear camera on some older iOS devices. I test on iPhone 11 Pro works, iPhone 7 does not work.
- My fix is ​​to update the code in the library: RNCamera.m change AVCaptureSessionPresetPhoto = AVCaptureSessionPresetHigh

```
-(AVCaptureSessionPreset)getDefaultPreset
{
    AVCaptureSessionPreset preset =
    ([self pictureSize] && [[self pictureSize] integerValue] >= 0) ? [self pictureSize] : **AVCaptureSessionPresetHigh**;

    return preset;
}
```

Everything works normally. I am not sure if there is any impact on react-native-camera functionality?

sibelius, 2020-07-22:
can you send a pull request?

PR #2193 (2019-04-11): [ios] migrate face detection to mlkit
Migrates face detection to MLKit,
Fixes #1870 #2033 #2100
Fixes #2183 - face detection is completely optional now via Cocoapods (which should be the [recommended way](https://facebook.github.io/react-native/blog/#leading-user-problems) now).
Should be fixing  #2184, as I was able to build basic example on XCode 10 after some changes in this PR (was an issue with incomplete RN upgrade on ios side)
closed 2020-09-27 by stale[bot]