react-native-camera/react-native-camera #1871: Could not find play-services-basement.aar (com.google.android.gms:play-services-basement:15.0.1).

AsocialCoder, 2018-10-23:
```
compile (project(':react-native-camera')) {
    exclude group: "com.google.android.gms"
    compile 'com.android.support:exifinterface:27.+'
    compile ("com.google.android.gms:play-services-vision:12.+")
}
```

Error: 

```
 Could not find play-services-basement.aar (com.google.android.gms:play-services-basement:15.0.1).
  Searched in the following locations:
      https://jcenter.bintray.com/com/google/android/gms/play-services-basement/15.0.1/play-services-basement-15.0.1.aar
```

iremlopsum, 2018-10-23:
Having the same problem and can't seem to find a solution. Updated `react-native-camera` from `1.1.0` to `1.3.0` and issue still exists

iremlopsum, 2018-10-23:
After Googling the hell out of this I found a solution based on this answer: https://stackoverflow.com/a/51213879/3840147

I changed my `build.gradle` files so that `jcenter()` would always be at the bottom of the list. I actually made this change for `react-native-camera` in `node_modules` as well and all works fine now :)

Seems that the library is down from maven's side, and needs to use a mirror instead

jiasongs, 2018-10-23:
same problem, +1

netdesk, 2018-10-23:
It was enough for me to move `jcenter()` below `google()` within `allprojects.repositories` in `android/build.gradle`. I don't know if this has side effects in other apps or configurations. At least using just this as a temporary solution you don't have to mess around in node_modules.

andikaerwansyah, 2018-10-23:
same problem, +1

xenu256, 2018-10-23:
Yeah, jcenter() after google() fixed the problem for me too.

sibelius, 2018-10-23:
fixed on master

closed 2018-10-23 by sibelius

SiSa68, 2018-11-11:
I'm using `react-native-firebase` and I had same problem, but non of the above answers works for me.
Updated `react-native-firebase`  and everything works fine now

imansalhi, 2018-11-28:
I'v same problem and the interesting thing is that i'v this problem only in macintosh ...
every thing is working when i build it on windows