react-native-camera/react-native-camera #1872: Could not find intellij-core.jar (com.android.tools.external.com-intellij:intellij-core:26.0.1).

Zabi-Progton, 2018-10-23:
When looking into the URL given it gives Resource not found error. The Jar File might have been removed. 

Steps to reproduce
1. React native init NewCameraProject
2. npm install react-native-camera --save
3. react-native link
4. react-native run-android

On executing step 4, it starts resolving the dependencies for ReactNative Camera: This error is shown in the Terminal:

* What went wrong:
A problem occurred configuring project ':react-native-camera'.
> Could not resolve all files for configuration ':react-native-camera:classpath'.
   > Could not find intellij-core.jar (com.android.tools.external.com-intellij:intellij-core:26.0.1).
     Searched in the following locations:
         https://jcenter.bintray.com/com/android/tools/external/com-intellij/intellij-core/26.0.1/intellij-core-26.0.1.jar

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

ger86, 2018-10-23:
This seems to work:

Change my build.gradle files so that jcenter() would always be at the bottom of the list. I actually made this change for react-native-camera in node_modules as well and all works fine now :)

Seems that the library is down from maven's side, and needs to use a mirror instead

Peretz30, 2018-10-23:
@ger86 can you make a merge request?

ger86, 2018-10-23:
Sure

sospedra, 2018-10-23:
Fix is coming: https://github.com/react-native-community/react-native-camera/pull/1874

ger86, 2018-10-23:
> Fix is coming: #1874

I was doing the fix....

satyen-progton, 2018-10-23:
former error is solved then i got this-
* What went wrong:
Could not resolve all files for configuration ':react-native-camera:debugCompileClasspath'.
> Could not download core.jar (com.google.zxing:core:3.3.0)
   > Could not get resource 'https://jcenter.bintray.com/com/google/zxing/core/3.3.0/core-3.3.0.jar'.
      > Read timed out

closed 2018-10-23 by sibelius

LostInReact, 2018-10-23:
@sospedra 
hello,
i have the same issue even after the fix :/
opérations :
rm -R node_modules
npm install git+https://github.com/react-native-community/react-native-camera.git
react-native run-android

PR #1874 (2018-10-23): Do not rely in to jcenter but google
Fixes #1872