react-native-camera/react-native-camera #1873: error: cannot find symbol variable surface_view

aman5537jains, 2018-10-23:
I am trying to compile my code after adding react native camera and i am getting below error

```
J:\react\Cropshop\node_modules\react-native-camera\android\src\main\java\com\google\android\cameraview\SurfaceViewPreview.java:34: error: cannot find symbol
        final View view = View.inflate(context, R.layout.surface_view, parent);
                                                        ^
  symbol:   variable surface_view
  location: class layout
J:\react\Cropshop\node_modules\react-native-camera\android\src\main\java\com\google\android\cameraview\SurfaceViewPreview.java:35: error: cannot find symbol
        mSurfaceView = (SurfaceView) view.findViewById(R.id.surface_view);
                                                           ^
  symbol:   variable surface_view
  location: class id
J:\react\Cropshop\node_modules\react-native-camera\android\src\main\java\com\google\android\cameraview\TextureViewPreview.java:38: error: cannot find symbol
        final View view = View.inflate(context, R.layout.texture_view, parent);
                                                        ^
  symbol:   variable texture_view
  location: class layout
J:\react\Cropshop\node_modules\react-native-camera\android\src\main\java\com\google\android\cameraview\TextureViewPreview.java:39: error: cannot find symbol
        mTextureView = (TextureView) view.findViewById(R.id.texture_view);
                                                           ^
  symbol:   variable texture_view
  location: class id
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: J:\react\Cropshop\node_modules\react-native-camera\android\src\main\java\com\google\android\cameraview\Camera2.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
```
Version : 

 react-native-camera  1.3.0
React native version 0.55.2
React 16.4.1
compileSdkVersion 27
buildToolsVersion 27.0.2

n1ru4l, 2018-11-20:
Closed due to inactivity

closed 2018-11-20 by n1ru4l

jbcullis, 2019-01-31:
Same issue here.

n1ru4l, 2019-01-31:
@jbcullis Please open a new issue with the issue template.