react-native-camera/react-native-camera #1875: Could not find intellij-core.jar error

LostInReact, 2018-10-23:
hello,
i have the same issue even after the fix #1874
opérations :
rm -R node_modules
npm install git+https://github.com/react-native-community/react-native-camera.git
react-native run-android

_Originally posted by @LostInReact in https://github.com/react-native-community/react-native-camera/issues/1872#issuecomment-432260690_

rikantro, 2018-10-23:
resolved with 
https://github.com/react-native-community/react-native-camera/issues/1871#issuecomment-432154734

4E71-NOP, 2018-10-23:
I got this on a fresh install and an up to date Android Studio:

```sh
WARNING: Configuration 'compile' is obsolete and has been replaced with 'implementation' and 'api'.
It will be removed at the end of 2018. For more information see: http://d.android.com/r/tools/update-dependency-configurations.html
WARNING: The specified Android SDK Build Tools version (26.0.1) is ignored, as it is below the minimum supported version (27.0.3) for Android Gradle Plugin 3.1.4.
Android SDK Build Tools 27.0.3 will be used.
To suppress this warning, remove "buildToolsVersion '26.0.1'" from your build.gradle file, as each version of the Android Gradle Plugin now has a default version of the build tools.

(...)

> Could not find intellij-core.jar (com.android.tools.external.com-intellij:intellij-core:26.0.1).
    Searched in the following locations:
        https://jcenter.bintray.com/com/android/tools/external/com-intellij/intellij-core/26.0.1/intellij-core-26.0.1.jar
```
I see "**removed at the end of 2018**", "**below the minimum supported version (27.0.3)**".

Maybe this module should not rely on a specific gradle/sdk version. As soon as something change we're back to square one (in a fresh install). I bet a lot of things are changing lately. It won't get better as Android SDK versions go up.

A lot if not all solutions i found are more like 'shaman rain dance' than proper configuration.

Anybody has a 100% sure shot fix ?

AndriiChubariev, 2018-10-23:
This error happened today not only with `react-native-camera`, but with the majority of RN libraries as well.

The solution is:
https://stackoverflow.com/questions/50563407/could-not-find-play-services-basement-aar/51213879#51213879

UPD: Also, for me, the `/node_modules/react-native-camera/android/build.gradle` files required update too.

change:
```
buildscript {
  repositories {
    jcenter()
    google()
    maven {
      url 'https://maven.google.com'
    }
    }
  }
```

to:
```
buildscript {
  repositories {
    maven {
    google()
    maven {
      url 'https://maven.google.com'
    }
   jcenter() // moved this line to the bottom.
  }
...
```

van1985, 2018-10-23:
When will be a new release for fix this issue? Thanks!!

ptomasroos, 2018-10-24:
This is fixed in master.

Feel free to update the build.gradle to force this to all projects in your local app build to fix this across all build.gradle dependencies.

```
allprojects {
    repositories {
        maven {
          url "https://maven.google.com"
          name 'Google'
        }
        jcenter()
    }
}
```

closed 2018-10-24 by ptomasroos

chaileicsu, 2018-10-24:
I move jcenter,like this:
repositories {
        google()
        maven {
                url 'https://maven.google.com'
        }
        jcenter()
 }

but 

`* What went wrong:
A problem occurred configuring project ':react-native-camera'.
> Could not resolve all files for configuration ':react-native-camera:classpath'.
   > Could not find intellij-core.jar (com.android.tools.external.com-intellij:intellij-core:26.0.1).
     Searched in the following locations:
         https://jcenter.bintray.com/com/android/tools/external/com-intellij/intellij-core/26.0.1/intellij-core-26.0.1.jar
`