react-native-camera/react-native-camera #1877: Text recognition for iOS without Cocoapods

ioanamirela, 2018-10-23:
# Feature-Request

**Describe the Feature**
Would love to use text recognition on iOS without the cocoapods constraint.

**Possible Implementations**

**Related Issues**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

sibelius, 2018-10-23:
you can use postinstall strategy

ioanamirela, 2018-10-23:
Do you mind clarifying?

jgcmarins, 2018-10-24:
@ioanamirela, check this: https://github.com/react-native-community/react-native-camera#face-detection-or-text-recognition-steps
you will find few instructions and a script to install text recognition without Cocoapods

ioanamirela, 2018-10-24:
Oh hey, there's a new "Non-CocoaPods Path" section. Thanks @sibelius, will try it out!

closed 2018-10-24 by ioanamirela