react-native-camera/react-native-camera #1879: iOS build fails from master with no FaceDetection

alan-nakagiri, 2018-10-24:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**

RNCamera
FaceDetector

**Platforms**

iOS

**Versions**

- Android:
- iOS: 12.0.0
- react-native-camera: 1.3.0
- react-native:0.57
- react: 16.5.1

**Description/Current Behaviour**
- v1.3.0 of react-native-camera was breaking earlier today while building Android. iOS was ok. 
-  #1874 fixed Android build but now iOS is now failing

- Building locally works just fine. The build fails when using circleci. 

**Expected Behaviour**

builds passes for both Android and iOS

**Steps to Reproduce**
fork off master and use in package.json
remove node modules, yarn install
relink RNCamera and remove FaceDetection in Xcode
deploy via circle ci
**build fails iOS deploy**

**Additionals**

 Looking at my circleci logs like it may have something to do with a change made from this [commit](https://github.com/react-native-community/react-native-camera/commit/1ecef00048fe0cea95dac0901309222b37ec2279#diff-deff8cbb3789a329a4a17e2e3bd13bce)

All of the failure come from that file at each of the lines changed in that commit where `<GoogleMobileVision/GoogleMobileVision.h>` was changed to `RNFaceDetectorManager.h`

I am getting the error `no known instance method for selector 'setIsEnabled:` or any of the other selectors used with `RNFaceDetectorManager.h`.

I did notice there is config options to run a postinstall script to remove FaceDetection and update the pbxproj so maybe I need to do that. But I did not need to change my build config when using v1.3.0 without FaceDetection. For that reason, I am posting this as a bug. 

Thanks!

alan-nakagiri, 2018-10-24:
Update: I picked reverted `RNFaceDetectorManager.h` to back to `<GoogleMobileVision/GoogleMobileVision.h>` and my build passes and the camera works 🤷‍♂️

brightsider, 2018-10-24:
Same issue

sypl, 2018-10-24:
Also facing this issue.

sibelius, 2018-10-24:
anybody can send a PR fixing this?

PR #1882 (2018-10-24): revert: check if face or text detector libraries are included independently
should fix #1879
closed 2018-10-24 by sibelius