react-native-camera/react-native-camera #1880: Display corners around detected bar code

zedtux, 2018-10-24:
# Feature-Request

**Describe the Feature**
Show corners, around the detected bar code, while scanning it, like is doing iOS in this screenshot :

[image]

SimonErm, 2018-11-08:
the barcode read event already includes the barcode bounds

n1ru4l, 2018-11-12:
An example/recipe for this could be provided in the documentation.

zedtux, 2019-06-01:
Could anyone add an example before this stupid bot wrongly close this issue?

MateusAndrade, 2019-06-01:
You can try: https://github.com/shahnawaz/react-native-barcode-mask

MateusAndrade, 2019-06-03:
#1503

zedtux, 2019-08-02:
I'm working on a big project right now stupid bot so go away, there are plenty of issues where you can bore people.

We have to upgrade RN in our app anyways due to the 64bits stuff so I'll give it a try at that moment.

zedtux, 2019-10-01:
🤫

zedtux, 2019-11-30:
😤

zedtux, 2020-01-30:
...

zedtux, 2020-03-30:
......

ceefour, 2020-04-12:
Related to: #874, #2786

ceefour, 2020-04-12:
Are the bounds accurate?

Example of how erratic the returned bounds are: (#2791)

https://www.youtube.com/watch?v=EGX49nYzqoQ

renatobenks, 2020-06-12:
@sibelius can we close that? it seems that this feature was introduced already, is that right?

closed 2020-06-12 by sibelius