react-native-camera/react-native-camera #1884: RN build failing because of a missing intellij-core library

van1985, 2018-10-24:
Hi,

react-native-camera: v1.1.4
RN version: 0.55.2

Notice: Same exact configuration was working yesterday (my colleagues with the jar in cache can still build the app)

Since this morning my build are failing because of a missing jar on jcenter.bintray.com (the pom file is there but nor the jar file):

FAILURE: Build failed with an exception.
* What went wrong:
A problem occurred configuring project ':react-native-camera'.
> Could not resolve all artifacts for configuration ':react-native-camera:classpath'.
   > Could not find intellij-core.jar (com.android.tools.external.com-intellij:intellij-core:26.0.1).
     Searched in the following locations:
         https://jcenter.bintray.com/com/android/tools/external/com-intellij/intellij-core/26.0.1/intellij-core-26.0.1.jar

The same happen to me with react-native-touch-id and make a fix.

I saw the fix in your commit but not in a new release. The commit is: https://github.com/react-native-community/react-native-camera/commit/9261524609ac4d9f29a33fb94847ef78a50b38f3

When do you think make a new release to fix this issue? 

Thanks in advance!
Leandro

sibelius, 2018-10-24:
use master or latest release

closed 2018-10-24 by sibelius

van1985, 2018-10-24:
@sibelius the commit is a day ago (https://github.com/react-native-community/react-native-camera/commit/9261524609ac4d9f29a33fb94847ef78a50b38f3) and the last release doesn't have this chante (release v1.3.0-8).

The only change is use master?

sibelius, 2018-10-24:
test new master, and if the fix works we can release a new version

tonmanayo, 2018-10-25:
Please release this