react-native-camera/react-native-camera #1888: autoFocusPointOfInterest not implemented in RNCamera

msageryd, 2018-10-25:
I tried to implement autoFocusPointOfInterest in my app, but it didn't work.

After digging into the RNCamera code I found that this prop isn't implemented at all. I could give it a stab, but I have never worked with ObjC before. If someone could give me some hints on how to send JS objects over the bridge I might be able to pull this off.

msageryd, 2018-10-25:
Oh, I'm sorry. I looked in a project which had RNCamera 1.20. autoFocusPointOfInterest was implemented in 1.30.

jasan-s, 2019-01-06:
@msageryd  how are you getting the coordinate values to use in `autoFocusPointOfInterest`. can you share the implementation gist

trinhlk31, 2019-07-16:
@msageryd Could you please share the code of "renderFocusPointOfInterest" function? Many thanks!

ShawalAhmadPF, 2020-08-24:
for renderFocusPointOfInterest
you can apply :
```
renderFocusPointOfInterest =() =>{
        return <View style={{ position:'absolute', top:touchPosition.x*Imageheight, left:touchPosition.y*ImageWidth, height:20, width:20, borderWidth:1, borderColor:'#fff' }} ></View>
}
```

closed 2021-01-13 by stale[bot]