react-native-camera/react-native-camera #1889: Sort out the zoom handling on iOS

msageryd, 2018-10-25:
I now realise that I have been writing in a closed issue. IMO this needs to be solved. I'm willing to help but I need some guidance.

Rather than linking to my posts in a closed issue I'll copy the text into this issue.

msageryd, 2018-10-25:
What's the argument for using a `0 to 1` zoom range? If I produce the zoom value from a pinch event I'll have to accumulate the value by multiplying the pinch ratio with the last known scale, i.e. I'll get numbers ranging from 1 and up. 

So, I have to convert my `1 to max` range to the `0 to 1` range.
RNCamera then converts the `0 to 1` range back to a `1 to max` range, since this is what ios wants.

**Reference docs:** https://developer.apple.com/documentation/avfoundation/avcapturedevice/1624611-videozoomfactor

Why not use an ios compliant number from start to finish, i.e. RNCamera.m would look like this (old solution remarked). N.B. this is my very first line of Objc code, ever. Be nice with me, Objc is terrifying..

According to the docs, videoZoomFactor must be within **1** and **videoMaxZoomFactor** or there will be an exception. 

```
- (void)updateZoom {
    AVCaptureDevice *device = [self.videoCaptureDeviceInput device];
    NSError *error = nil;

    if (![device lockForConfiguration:&error]) {
        if (error) {
            RCTLogError(@"%s: %@", __func__, error);
        }
        return;
    }

    device.videoZoomFactor = MIN(MAX(self.zoom, 1), device.activeFormat.videoMaxZoomFactor);

    //device.videoZoomFactor = (device.activeFormat.videoMaxZoomFactor - 1.0) * self.zoom + 1.0;

    [device unlockForConfiguration];
}
```

**EDIT:**
A big problem with the current solution (appart from it being broken) is that there is no way of matching the zoom scale with the pinch action. When the user doubles the distance between the two fingers a zoom of 200% is expected. This can be achieved with my proposed solution. 

The current solution transposes the zoom values to an unknown zoom range, since the rescaling is device specific (videoMaxZoomFactor differs between devices).

If RNCamera maintainers wants to keep the `0 to 1` zoom range we need to get hold of `videoMaxZoomFactor` to be able to adjust for the secret calculations.

msageryd, 2018-10-25:
I have tried to dig in some more in RNCamera to try to solve my task.

- I really need to get hold of `videoMaxZoomFactor`. 

- I would also like to get hold of `videoZoomFactorUpscaleThreshold` so I can know if the optical zoom is used or if I'm just cropping the image. Normally my app captures medium sized images. But if I knew that the image was digitally zoomed (cropped) I'd bump up the output resolution.

How can we get hold of those parameters? Best would be if they could be delivered to the JS side as static functions on the RNCamera object. But I couldn't get this to work since the captureDevice isn't yet initialised. Further more, the parameters must be fetched from the correct device (front camera, back camera, video or still, etc).

Could we have a "deviceInfo" object sent via the onCameraReady event? This would be great IMO, but I have barely seen ObjC before, so I'd have much trouble to get this together.

If someone could implement this (or guide me) we would't need to alter the `updateZoom()` function as per the previous post, which would be a breaking change. Instead we could solve the zoom range problem via some simple calculations.

closed 2019-06-08 by stale[bot]

lorissikora, 2023-06-04:
@msageryd Have you ever been able to work around this issue? I've been experiencing similar issues and have no solution in sight...