react-native-camera/react-native-camera #1895: Grab frames from Camera

CPIJ, 2018-11-01:
Hi!

I'm trying to make an object classifier using React Native. Just starting to explore this library, but I was wondering, is it possible to get a live feed from RNCamera with each frame as a separate image?

Some pseudocode to illustrate my question:

```javascript
render() {
 <RNCamera
    ref={ref => {
      this.camera = ref;
    }}
    onFrame={processFrame}
 />
}

processFrame(image) {
 // Get the predictions from the image classifier and do something useful with them.
}
```

sibelius, 2018-11-01:
you can create an new `event` here https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraView.java#L119

so you can send the data to javascript at each frame

this won't be so performatic

or you can try to explore this https://github.com/react-native-community/react-native-camera/issues/1262

where you can create a native code that could be used on native side

CPIJ, 2018-11-01:
Thanks sibelius, I'm not looking for the best performance. Just making a small proof of concept :)

Manthika, 2019-05-11:
> I tried implementing this in my project to recognize fruits from camera preview in real time using TensorFlow. Instead of grabbing the frames and converting it to jpeg format or base64 string and passing it to javascript, I grabbed the frames via TextureView.getBitmap() from CameraView and feed it to tensorflow for image recognition and passed the results to Javascript. It's fast and could process frames in real time. If you're going to convert the byte[] data from onFramePreview to other format, it may take a while to pass it to javascript.
> 
> Workflow for Real Time Image Processing:
> Slow and Inefficient -
> (Native) Grab Frames -> (Bridge) Pass Frames -> (Javascript) Receive Frames -> (Bridge) Send Frames for Processing -> (Native) -> Process Frames and get the results-> (Bridge) Pass Results -> (Javascript) Get Results
> 
> Fast and Efficient -
> (Native) Grab frames and process the frames immediately(If you're dealing with bitmaps, I suggest to use TextureView.getBitmap() in working with frames instead of converting byte data[] to other formats such as jpeg) -> (Bridge) Pass Results -> (Javascript) Receive Results.

Could you please explain a bit more

SNWStaff, 2019-09-17:
> I tried implementing this in my project to recognize fruits from camera preview in real time using TensorFlow. Instead of grabbing the frames and converting it to jpeg format or base64 string and [...]

Hello people, can I get more information about this technique please?

shreyakamoji96, 2019-09-25:
> I tried implementing this in my project to recognize fruits from camera preview in real time using TensorFlow. Instead of grabbing the frames and converting it to jpeg format or base64 string and [...]

Heyy can we get more explanation or any sort of example applications

jslok, 2019-11-08:
@jazzystring1 I'm also interested in your faster method. If you can point us to what file/where to grab the frames like that it would be much appreciated!

Papaass, 2020-01-01:
Hello,
Can you share the code to do that?
Thx

rafaelcavalcante, 2020-09-29:
Hi! Any progress on this?

closed 2021-01-13 by stale[bot]

edritech93, 2021-07-05:
> I tried implementing this in my project to recognize fruits from camera preview in real time using TensorFlow. Instead of grabbing the frames and converting it to jpeg format or base64 string and [...]

need more info

rafaelcavalcante, 2021-07-05:
> need more info

Please be more specific about what information you need