react-native-camera/react-native-camera #1897: RNCamera [android] recordAsync method doesn't work

zouminhuababa, 2018-11-02:
Related Modules
RNCamera

Platforms
Android

Versions

android:
react-native-camera: ^1.3.1
react-native: 0.57.4
react: 16.6.0-alpha.8af6728

when i press record button i got this :

 Possible Unhandled Promise Rejection (id: 0):
Error: setAudioSource failed.
Error: setAudioSource failed.
 
I have added all the necessary permissions。
  <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.RECORD_AUDIO"/>
    <uses-permission android:name="android.permission.RECORD_VIDEO"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

my code : 

import React, { Component } from 'react';
import { StyleSheet, Text, TouchableOpacity, View } from 'react-native';
import { RNCamera } from 'react-native-camera';

class CameraRecordScreen extends Component {
    constructor(props){
        super(props);
        this.state={
            isFlashOn:false,
            isRecording:false,
        }

    }

    componentDidMount() {
    };

    render() {
        return (
            <View style={styles.container}>
                <RNCamera
                    ref={ref => {
                        this.camera = ref;
                    }}
                    style={styles.preview}
                    type={RNCamera.Constants.Type.back}
                    flashMode={RNCamera.Constants.FlashMode.off}
                    permissionDialogTitle={'Permission to use camera'}
                    permissionDialogMessage={'We need your permission to use your camera phone'}
                >
                    <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
                        <View style={{position:'absolute',top:30,left:-40}} />
                        {this.state.isRecording ?
                            <TouchableOpacity
                                onPress={this.stopRecord.bind(this)}
                                style={styles.capture}>
                                <Text style={{ fontSize: 14 }}> Stop </Text>
                            </TouchableOpacity> :
                            <TouchableOpacity
                                onPress={this.takeRecord.bind(this)}
                                style={styles.capture}>
                                <Text style={{ fontSize: 14 }}> Record </Text>
                            </TouchableOpacity>
                        }
                    </View>
                </RNCamera>
            </View>
        );
    }

    takeRecord= async function(){
        if (this.camera) {
            this.setState({isRecording:true});
            const options = { quality:RNCamera.Constants.VideoQuality["480p"],maxFileSize:(100*1024*1024), maxDuration: 60 };
            const data = await this.camera.recordAsync(options);
            console.log('takeRecord data', data);
        }
    };

    stopRecord = () => {
        if (this.camera) {
            this.setState({isRecording:false});
            this.camera.stopRecording()
        }
    }
}

const styles = StyleSheet.create({
    container: {
        flex: 1,
        flexDirection: 'column',
        backgroundColor: 'black',
    },
    preview: {
        flex: 1,
        justifyContent: 'flex-end',
        alignItems: 'center',
    },
    capture: {
        flex: 0,
        backgroundColor: '#fff',
        borderRadius: 5,
        padding: 15,
        paddingHorizontal: 20,
        alignSelf: 'center',
        margin: 20,
    },
});

export default CameraRecordScreen

closed 2018-11-02 by zouminhuababa

AlexSeoven, 2018-11-17:
How do you solve your problem?

swaroopa94, 2018-11-19:
@zouminhuababa Can you tell me how did you solve this problem..

xinggangling, 2018-11-22:
I have the same problem;

"react-native-camera": "1.4.3",
 <uses-permission android:name="android.permission.CAMERA"/>
<uses-permission android:name="android.permission.RECORD_AUDIO"/>
    <uses-permission android:name="android.permission.RECORD_VIDEO"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
ext {
    buildToolsVersion = "26.0.3"
    minSdkVersion = 16
    compileSdkVersion = 26
    targetSdkVersion = 26
    googlePlayServicesVersion = "12.0.1"
    supportLibVersion = "27.1.0"
}

Error: setAudioSource failed.
when I click recordAsync, simulator crash;
[image]

😫😫😫

n1ru4l, 2018-11-26:
@xinggangling Does it also crash on a real device?

icesyc, 2018-12-26:
set the prop  
```
captureAudio={true}
```
fixed the problem.

zouminhuababa, 2018-12-26:
> @zouminhuababa 你好，我也遇到了这个问题，方便说一下是怎么解决的这个问题吗

给应用的麦克风权限授权

n1ru4l, 2018-12-26:
Please write english to help everyone. I am going to lock conversation. Recording does only work on real devices..