react-native-camera/react-native-camera #1900: Incompatible AVCAptureSessionPreset causes crash on iPad Mini 4

kjartanvest, 2018-11-02:
**Versions**
react-native-camera: 1.3.1
react-native: 0.44
react: 16.0.0-alpha.6
Platform: iOS 11/iOS 12

**Description/Current Behaviour**
Using the latest release on master (1.3.1), opening the camera causes the app to crash when using an iPad Mini 4. This error gets printed in Xcode:

*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCaptureSession setSessionPreset:] AVCaptureSessionPreset3840x2160 is not a supported preset'

Seems like this happens because RNCamera is trying to use the highest resolution video preset (4K video), but the camera in the iPad Mini does not support 4K video.

**Expected Behaviour**
The camera should start without crashing.

**Additionals**
In my package.json: "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera",
I am using RNCamera without the FaceDetector folder.

cozmo, 2018-11-07:
I'm also experiencing this on both an iPhone XS with iOS 12 and a 5th generation iPad with iOS 11. Oddly for both of these devices this issue has started without any changes to the underlying package (version locked on commit 537f4bedf0f8172b4f96cd0dd997286ba5b497e2). Unclear what triggered this.

closed 2018-11-07 by n1ru4l

n1ru4l, 2018-11-12:
:tada: This issue has been resolved in version 1.4.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.4.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #1907 (2018-11-07): Add the types for defaultVideoQuality
In a half hearted attempt to fix #1900 I wanted to try to configure the default video quality using the new prop (`defaultVideoQuality`) that was introduced in #1886, but it's not defined in the typescript definition file. 

This PR just adds the types. Should have no impact on anything since it's just a change to the type definitions, so as long as it seems to correctly define the API was added in #1886.
PR #1908 (2018-11-07): Fix default resolution on iOS
I believe this is the fix to #1900, fixing a bug that was introduced in #1886. 

The root of the issue is that [this enum](https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCameraManager.h#L50-L56) for video resolution defines `0` as `RNCameraVideo2160p`, and `defaultVideoQuality` was an Integer, so if it was not set it was mapping to `0` and thus to `RNCameraVideo2160p`, causing any user of this library who didn't pass in an explicit `defaultVideoQuality` to request `AVCaptureSessionPreset3840x2160` which is far higher than many i [... cut]
commit 41079d54 (2018-11-07): fix(ios): select correct capture session preset when defaultVideoQuality is not provided  (#1908)

Closes #1900