react-native-camera/react-native-camera #1902: How make RNCamera to fit to container on Android?

seraphmate, 2018-11-05:
# Question

**To Do First**
- [O] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [O] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [O] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

Firstly, I really thank for all developers who keep this useful library. :)
And, I'm sorry about I can't explain enough cause my english is too short.
I upgraded RNCamera(RCTCamera) from 0.12.0 to master, and I just use camera preview for barcode scanning.
When I use RCTCamera, RCTCamera was fit to it's container, but after upgrade library, RNCamera is not to fit container on Android.
RNCamera show over it's container. How can I make RNCamera to fit container?
I want to make my code like iOS Screenshot.

** Screenshot on Android **
[image]

** Screenshot on iOS **
[image]

```javascript
<View style={{flex: 1, flexDirection: 'column', height: Utils.Size.convert(300)}}>
   <RNCamera
        style={styles.camera}
        onBarCodeRead={(code) => {
          if (this.state.barcode === '') this._onBarcodeRead(code)
        }}
        flashMode={this.props.flashOn ? RNCamera.Constants.FlashMode.torch : RNCamera.Constants.FlashMode.off}
        captureAudio={false}
        showViewFinder={false}>
        {scanArea}
      </RNCamera>
</View>
```

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

NSShentu, 2018-11-06:
and how can we hide the navigation bar in android？
i hide bar in mainActivity but it comes out when i go into camera page

maxmijn, 2018-11-07:
+1. After upgrading my buildToolsVersion to "27.0.3" and the compileSdkVersion to 27, the camera does not fit to the view anymore. Setting a fixed height, width or ratio does not work.

maxmijn, 2018-11-07:
@sibelius This works! Not sure what this helper actually does, but I got a square camera view again. Thanks.

seraphmate, 2018-11-12:
@sibelius Thank you for your guide. :)

closed 2018-11-12 by seraphmate

ReneDH, 2019-04-02:
Thank you so much - it's works perfectly

jaredramirez, 2019-07-24:
If anyone needs this in reason react native, i converted it:
```
module FillToAspectRatio = {
  open ReactNative;

  let styles =
    Style.(
      StyleSheet.create({
        "container":
          style(~flex=1., ~overflow=`hidden, ~position=`relative, ()),
      })
    );

  type layoutInfo = {
    width: float,
    height: float,
  };

  [@react.component]
  let make = (~ratio: (float, float)=(4., 3.), ~children: React.element=?) => {
    let (layoutInfo, setLayoutInfo) = React.useState(() => Js.Nullable.null);

    let handleLayout = layoutEvent => {
      let nativeEvent = layoutEvent##nativeEvent;
      setLayoutInfo(_ =>
        Js.Nullable.return({
          width: nativeEvent##layout##width,
          height: nativeEvent##layout##height,
        })
      );
    };

    let getRatio = () => {
      let (ratioWidth, ratioHeight) = ratio;
      ratioHeight /. ratioWidth;
    };

    switch (layoutInfo->Js.Nullable.toOption) {
    | None =>
      <View key="pre-info" onLayout=handleLayout style=styles##container />
    | Some({width, height}) =>
      let ratio = getRatio();
      let (wrapperWidth, wrapperHeight) =
        if (ratio *. height < width) {
          (width, width /. ratio);
        } else {
          (ratio *. height, height);
        };
      <View onLayout=handleLayout style=styles##container>
        <View
          style=Style.(
            style(
              ~width=wrapperWidth->dp,
              ~height=wrapperHeight->dp,
              ~marginLeft=((width -. wrapperWidth) /. 2.)->dp,
              ~marginTop=((height -. wrapperHeight) /. 2.)->dp,
              (),
            )
          )>
          children
        </View>
      </View>;
    };
  };
};
```

Sargnec, 2020-08-17:
Used this but after taken pictures becomes bigger. I think overflow: hidden solves nothing

johhansantana, 2022-07-06:
not sure if this helps but setting the container style `overflow: 'hidden'` on android fixes the overlay of the camera ratio.