react-native-camera/react-native-camera #1909: Android Text Recognize Error.

dileepbolisetti, 2018-11-07:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
I Was Trying To implement text Recognition Normal Camera is working fine When i Am adding this

           **onTextRecognized={({ textBlocks }) =>
              this.setState({ detectedTexts: textBlocks.map(b => b.value) })
            }**
I Am Having this Error... Apk build is Sucessfull
[image]

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

SimonErm, 2018-11-08:
did you modify your gradle files like discribed here:https://github.com/react-native-community/react-native-camera/blob/master/README.md ?

dileepbolisetti, 2018-11-11:
@SimonErm  thanks after adding few dependencies the error gone...

**implementation "com.google.zxing:core:3.2.1"
implementation "com.drewnoakes:metadata-extractor:2.9.1"**

How ever android Documentation is not as clear as ios.

closed 2018-11-11 by dileepbolisetti

n1ru4l, 2018-11-12:
@dileepbolisetti Feel free to improve the documentation for android!

g4laura, 2018-12-05:
I'm having the same error, would you mind sharing more about how you fixed it?

dileepbolisetti, 2018-12-06:
@g4laura 
[](https://github.com/react-native-community/react-native-camera/issues/1909#issuecomment-437650544)

sibelius, 2018-12-06:
we need more docs

guhyeon, 2018-12-17:
In my case, I have solved by
Adding jitpack to android/build.gradle
and gradle cache clean cd android && ./gradlew clean

//Adding jitpack to android/build.gradle
allprojects {
    repositories {
        maven { url "https://jitpack.io" }
        maven { url "https://maven.google.com" }
    }
}