react-native-camera/react-native-camera #1910: setPreviewTexture Failed

lothindil, 2018-11-07:
# Bug Report
**To Do First**
- "react-native-camera": "^1.2.0"
- "react-native-qrcode-scanner": "^1.0.1",

**Related Modules**
RNCamera / qrcode scanner

**Platforms**
Android 7

**Versions**
- Android: 7.12
- react-native-camera: 1.2.0
- react-native: 0.55.4
- react: 16.3.1

**Description/Current Behaviour**
Error "SetPreviewTexture" after read of qr code
https://drive.google.com/file/d/1BjTwd_zfrmTCwBHQaCxPyf8YME1cAHgU/view?usp=sharing

**Expected Behaviour**
Nothing, just go to next part of scene

**Steps to Reproduce**
My app' is conceived to use several times three times camera : 1 scan, 1 photo, 1 scan.

We must be able to do this several times.

We navigates screen to screen with modify state on a reducer. Basically our system is : 

(0) state : "Start" > welcome page (indexScene.js) > button "scan" > actionScene > reducer Scene
(1) state : "ItemScan" > scan qr code (scanScene.js) > actionScene > reducerScene
(2) state : "ItemForm" > form (scanScene.js) > actionScene > reducerScene
(3) state : "ItemPict" > photo  (pictureScene.js) > actionScene > reducerScene
(4) state : "WHScan" > scan qr code (scanScene.js) > actionScene > reducerScene

"continue" or "leave" button (scanScene.js) > actionScene > reducerScene 

If "continue" return to 1.
if "leave" return to 0.

leave + button scan = continue

The issue is in the beggining of step 2.

And don't depend about the choice.  I don't have any way to follow...

SimonErm, 2018-11-08:
Can you provide the code of your camera and scanner component?

lothindil, 2018-11-09:
For sure.

photoPage.js

    render()
    {
        return (
        
        <View style={styles.container}>
            <Text>Prendre une photo</Text>
          <RNCamera
              ref={ref => {
                this.camera = ref;
              }}
              style = {styles.preview}
              type={RNCamera.Constants.Type.back}
              flashMode={RNCamera.Constants.FlashMode.on}
              permissionDialogTitle={'Permission to use camera'}
              permissionDialogMessage={'We need your permission to use your camera phone'}
          />
          <View style={{flex: 0, flexDirection: 'row', justifyContent: 'center',}}>
          <TouchableOpacity
              onPress={this.takePicture.bind(this)}
              style = {styles.capture}
          >
              <Text style={{fontSize: 14}}> Photo de l'objet </Text>
          </TouchableOpacity>
          </View>
        </View>
      );
        
    }
    takePicture = async function() {
        const {wh} = this.props;
        if (this.camera) {
            const options = { quality: 0.75, base64: true, width:800, doNotSave:true };
            const data = await this.camera.takePictureAsync(options);
                const {wh} = this.props;
                console.log("DEBUG : Photo Page; takepicture; ref : "+wh.id);
                await this.props.warehouseAddPhoto(wh.id,data.base64);
                Actions.warehouse();
         }
    }
}

ScanPage.js

    _renderScan()
    {
        const {scan} = this.props;
        switch(scan.step)
        {
            case "c_waiting":
                titreScan="Scanner le Premier QR du Client";
                break;
            case "c_waitingSecond":
                titreScan="Scanner le dernier QR du Client";
                break;
        } 
        return(
            <Container>
                <Text>{titreScan}</Text>
                <QRCodeScanner
                    ref={(node) => { this.scanner = node }} 
                    onRead={this.onSuccess.bind(this)}
                    TopContent={<Text>{titreScan}</Text>}
                />
                <AwesomeAlert
                show={scan.scanShowAlert}
                showProgress={false}
                title="Erreur"
                message={scan.scanMessage}
                closeOnTouchOutside={true}
                closeOnHardwareBackPress={false}
                showConfirmButton={true}
                confirmText="Réessayer"
                confirmButtonColor="#DD6B55"
                contentContainerStyle={{paddingLeft: 3, paddingRight: 3}}
                onConfirmPressed={this.onAlertSeenPress.bind(this)}
                />
            </Container>
            
        );
    }

Fonctions associées

    onSuccess(e) {
        try
        {
            const {scan} = this.props;
            switch(scan.step)
            {
                case "c_waiting":
                    this.props.scanClientFirstItem(e.data);
                    break;
                case "c_waitingSecond":
                    this.props.scanClientLastItem(e.data);
                    break;
            }
        }
      }

    onAlertSeenPress()
    {
        console.log("DEBUG : Page Scan; onAlertSeenPress(),partie");
        this.props.scanAlertSeen();
        this.scanner.reactivate();
        
    }

lothindil, 2018-11-09:
Isolating the utilisation of module, we still have the issue.

In fact, the issue is more easier, we have the error when we use several times the camera, alone or with the module qrcode (who depend of the react-native-camera).

I asked myself why the error tell about "camera1" ? This modul is depreciated on Android since API 21, I work with API 28. Maybe the error come from there ?

sibelius, 2018-11-09:
Camera2 has some bugs, his module will fallback to camera1 if you don’t explicit set it to use camera 2 api

sraka1, 2019-06-21:
@sibelius could you re-open this issue please? It still occurs.

MateusAndrade, 2019-06-22:
@sraka1, are you going to something similar? You can provide a sample showing your code?

closed 2019-08-28 by stale[bot]