react-native-camera/react-native-camera #1914: Android app crashes sometimes when calling `takePictureAsync`

mconiglio, 2018-11-09:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
Android

**Versions**
- Android: 5.1.1
- react-native-camera: 1.3.1
- react-native: 0.55.4
- react: 16.3.1
- react-navigation: 2.17.0

**Description/Current Behaviour**
App crashes sometimes when calling `takePictureAsync`. This never happens on the emulator, I'm seeing it happen on a Samsung Galaxy J3. Taking the picture works randomly around 3 of 4 times, but sometimes it makes the app crash.

**Expected Behaviour**
It should take the picture and don't crash.

**Steps to Reproduce**
Taking a picture 4 o 5 times and it will fail.
```javascript
import React from 'react';
import { View, TouchableOpacity, Image } from 'react-native';
import { RNCamera } from 'react-native-camera';
import styles from './styles';
import refreshIcon from '../../assets/icons/refresh.png';

class SelectMediaScreen extends React.Component {
  constructor(props) {
    super(props);
    this.state = {
      imageData: null,
      cameraType: RNCamera.Constants.Type.back,
    }
  }

  takePicture = () => {
    if (this.camera) {
      const options = { quality: 0.5, base64: true, fixOrientation: true };
      this.camera.takePictureAsync(options).then(data => {
        const { uri, base64 } = data;
        this.setState({ imageData: { uri, base64 } });
      }).catch(console.log)
    }
  };

  handleCameraTypeChange = () => {
    const { back, front } = RNCamera.Constants.Type;
    this.setState({ cameraType: this.state.cameraType === back ? front : back });
  }

  render() {
    const { imageData, cameraType } = this.state;

    return (
      <View style={styles.container}>
        {imageData && (
          <View style={styles.content}>
            <Image style={styles.preview} source={{ uri: imageData.uri }} />
          </View>
        )}
        {!imageData && (
          <View style={styles.content}>
            <RNCamera
              ref={ref => { this.camera = ref; }}
              style={styles.preview}
              type={cameraType}
            >
              <TouchableOpacity style={styles.cameraTypeIconContainer} onPress={this.handleCameraTypeChange}>
                <Image source={refreshIcon} style={styles.cameraTypeIcon} />
              </TouchableOpacity>
            </RNCamera>
          </View>
        )}
        <View style={styles.actions}>
          <TouchableOpacity onPress={this.takePicture}>
            <View style={styles.cameraButton} />
          </TouchableOpacity>
        </View>
      </View>
    );
  }
};

export default SelectMediaScreen;
```

n1ru4l, 2018-11-12:
Would be beneficial if you cloud provide a stack trace.

Also please try using the master branch (https://github.com/react-native-community/react-native-camera#how-to-use-master-branch), you problem could already be fixed.

mconiglio, 2018-11-12:
Thanks @n1ru4l,

I tried using the master branch and this is happening there as well.

I haven't been able to get a stack trace because I couldn't connect this device to android studio. I was able to connect a Samsung A5 2017 but the error doesn't show up for that one.

I was thinking that for that reason it may be a something related to the hardware and the options I'm passing to the `takePictureAsync` method. Could that be possible? I tried using different options on the method call but I wasn't able to fix it (for Samsung Galaxy J3).

sibelius, 2018-11-12:
try to do `adb logcat`

mconiglio, 2018-11-13:
@sibelius I tried several times with that (disconnecting and re-connecting the device) but the device doesn't show up. I also disabled and re-enabled the usb debugging and the whole developer options. 

The only way I had to put the apk in the device is by sending it by bluetooth. But I'm still not able to make it work. I don't know if it's because of the device not supporting it or because of something I'm missing.

Nocturne-FFG, 2018-12-01:
Hey friend! Have you solved this problem? I am more serious, I will crash every time, never succeeded on Android.

mconiglio, 2018-12-03:
Hey @Nocturne-FFG, no I haven't. I tried different options and even setting the quality to 0.1, but nothing seems to work

brunotalhate, 2018-12-12:
I'm facing a similar case, but the app crashes every time takePictureAsync is called with base64 on, but without base64 it works properly.
Tested in a Samsung Galaxy C5 Pro. Working well on other android devices.

n1ru4l, 2018-12-12:
Please provide Crash Logs

AbsohAbsoh, 2018-12-19:
I'm also experiencing this issue regardless of base64 being on/off.

n1ru4l, 2018-12-19:
@AbsohAbsoh Please provide crashlogs.

benhartouz, 2018-12-25:
Same issues with me , and i got this stack 

```Android: 7.0
Android Build: NRD90M.A710FXXU2CQE3
Manufacturer: samsung
Model: SM-A710F
Thread: AsyncTask #5-11390
CrashReporter Key: 00850222-3538-410a-a855-4c14a35a6a65
Start Date: 2018-12-21T15:52:28.676Z
Date: 2018-12-21T15:53:31.725Z

java.lang.RuntimeException: An error occurred while executing doInBackground()
	at android.os.AsyncTask$3.done(AsyncTask.java:318)
	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
	at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
	at java.util.concurrent.FutureTask.run(FutureTask.java:242)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
	at java.lang.Thread.run(Thread.java:762)
Caused by: java.lang.OutOfMemoryError: Failed to allocate a 51121164 byte allocation with 16777216 free bytes and 38MB until OOM
	at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
	at android.graphics.Bitmap.nativeCreate(Native Method)
	at android.graphics.Bitmap.createBitmap(Bitmap.java:977)
	at android.graphics.Bitmap.createBitmap(Bitmap.java:948)
	at android.graphics.Bitmap.createBitmap(Bitmap.java:879)
	at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.rotateBitmap(ResolveTakenPictureAsyncTask.java:162)
	at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:92)
	at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:26)
	at android.os.AsyncTask$2.call(AsyncTask.java:304)
	at java.util.concurrent.FutureTask.run(FutureTask.java:237)```

benhartouz, 2018-12-25:
@mconiglio  I've just solved this problem by adding "android:largeHeap="true"" to android/src/main/AndroidManifest.xml. and rebuild your project with react-native run-android

for example:
...
android:name=".MainApplication"
android:allowBackup="true"
android:largeHeap="true"*
android:label="@string/app_name"
android:icon="@mipmap/ic_launcher"
android:theme="@style/AppTheme">

joshuafarray, 2018-12-26:
Same issue for me.  

Android version: 7.0
Android Build: NPN25.137-33
Manufacturer: Motorola
Model: Moto G5 Plus

I'm not sure how much is relevant, but this is what I see in logcat:

```
12-26 13:23:39.113  3467  3801 I art     : Starting a blocking GC Alloc
12-26 13:23:39.113  3467  3801 I art     : Starting a blocking GC Alloc
12-26 13:23:39.156  3467  3801 I art     : Alloc sticky concurrent mark sweep GC freed 3726(137KB) AllocSpace objects, 7(140KB) LOS objects, 3% free, 160MB/167MB, paused 1.876ms total 41.840ms
12-26 13:23:39.156  3467  3801 I art     : Starting a blocking GC Alloc
12-26 13:23:39.200  3467  3801 I art     : Alloc partial concurrent mark sweep GC freed 2320(81KB) AllocSpace objects, 0(0B) LOS objects, 9% free, 160MB/176MB, paused 1.285ms total 43.225ms
12-26 13:23:39.200  3467  3801 I art     : Starting a blocking GC Alloc
12-26 13:23:39.242  3467  3801 I art     : Alloc concurrent mark sweep GC freed 287(13KB) AllocSpace objects, 0(0B) LOS objects, 9% free, 160MB/176MB, paused 1.238ms total 41.930ms
12-26 13:23:39.243  3467  3801 I art     : Forcing collection of SoftReferences for 34MB allocation
12-26 13:23:39.243  3467  3801 I art     : Starting a blocking GC Alloc
12-26 13:23:39.286  3467  3801 I art     : Alloc concurrent mark sweep GC freed 51(1528B) AllocSpace objects, 0(0B) LOS objects, 9% free, 160MB/176MB, paused 1.430ms total 43.188ms
12-26 13:23:39.286  3467  3801 W art     : Throwing OutOfMemoryError "Failed to allocate a 36578316 byte allocation with 16777216 free bytes and 31MB until OOM"
12-26 13:23:39.286  3467  3801 I art     : Starting a blocking GC Alloc
12-26 13:23:39.286  3467  3801 I art     : Starting a blocking GC Alloc
12-26 13:23:39.294  3467  3801 I art     : Alloc sticky concurrent mark sweep GC freed 4(432B) AllocSpace objects, 0(0B) LOS objects, 9% free, 160MB/176MB, paused 1.185ms total 7.413ms
12-26 13:23:39.294  3467  3801 I art     : Starting a blocking GC Alloc
12-26 13:23:39.330  3467  3801 I art     : Alloc partial concurrent mark sweep GC freed 217(7KB) AllocSpace objects, 0(0B) LOS objects, 9% free, 160MB/176MB, paused 1.477ms total 35.702ms
12-26 13:23:39.330  3467  3801 I art     : Starting a blocking GC Alloc
12-26 13:23:39.376  3467  3801 I art     : Alloc concurrent mark sweep GC freed 1395(42KB) AllocSpace objects, 0(0B) LOS objects, 9% free, 160MB/176MB, paused 1.895ms total 45.882ms
12-26 13:23:39.376  3467  3801 I art     : Forcing collection of SoftReferences for 34MB allocation
12-26 13:23:39.376  3467  3801 I art     : Starting a blocking GC Alloc
12-26 13:23:39.424  3467  3801 I art     : Alloc concurrent mark sweep GC freed 1869(58KB) AllocSpace objects, 0(0B) LOS objects, 9% free, 160MB/176MB, paused 2.212ms total 47.208ms
12-26 13:23:39.424  3467  3801 W art     : Throwing OutOfMemoryError "Failed to allocate a 36578316 byte allocation with 16777192 free bytes and 31MB until OOM"
12-26 13:23:39.454  3467  3801 D skia    : Error: Could not allocate pixels.
12-26 13:23:39.454  3467  3801 D skia    : --- Failed to decode region.
12-26 13:23:39.457  3467  3801 E AndroidRuntime: FATAL EXCEPTION: AsyncTask #5
12-26 13:23:39.457  3467  3801 E AndroidRuntime: Process: com.myapp, PID: 3467
12-26 13:23:39.457  3467  3801 E AndroidRuntime: java.lang.RuntimeException: An error occurred while executing doInBackground()
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at android.os.AsyncTask$3.done(AsyncTask.java:318)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:242)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:761)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: Caused by: java.lang.OutOfMemoryError: Failed to allocate a 36578316 byte allocation with 16777192 free bytes and 31MB until OOM
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at android.graphics.BitmapRegionDecoder.nativeDecodeRegion(Native Method)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at android.graphics.BitmapRegionDecoder.decodeRegion(BitmapRegionDecoder.java:188)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at com.facebook.react.modules.camera.ImageEditingManager$CropTask.crop(ImageEditingManager.java:306)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at com.facebook.react.modules.camera.ImageEditingManager$CropTask.doInBackgroundGuarded(ImageEditingManager.java:274)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at com.facebook.react.modules.camera.ImageEditingManager$CropTask.doInBackgroundGuarded(ImageEditingManager.java:203)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at com.facebook.react.bridge.GuardedAsyncTask.doInBackground(GuardedAsyncTask.java:32)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at com.facebook.react.bridge.GuardedAsyncTask.doInBackground(GuardedAsyncTask.java:20)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at android.os.AsyncTask$2.call(AsyncTask.java:304)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:237)
12-26 13:23:39.457  3467  3801 E AndroidRuntime: 	... 3 more
12-26 13:23:39.460  1549  6009 W ActivityManager:   Force finishing activity com.myapp/.MainActivity
```

`android:largeHeap="true"` is set in my android manifest.

n1ru4l, 2018-12-26:
@benhartouz @joshuafarray The only solution here would be to reduce memory consumption during the background task (which does some processing to the image).

There is some task that can require huge amounts of memory being done in the following lines here: https://github.com/react-native-community/react-native-camera/blob/5e5da25276e031b3b1616d1cb2acd980ce6459b5/android/src/main/java/org/reactnative/camera/tasks/ResolveTakenPictureAsyncTask.java#L83

Related: https://github.com/react-native-community/react-native-camera/issues/1053 (https://github.com/react-native-community/react-native-camera/issues/1053#issuecomment-448658325)

If you do not need them you can try to use the `skipProcessing` option for `takePictureAsync`. However this should only be a temporary solution.

It would be awesome if any of you could investigate more on a solution of this issue. A first step would be to compare how https://github.com/CameraKit/camerakit-android does image processing and whether it is less memory consumption intense or not. The next step could be a pull request that utilizes the same library that `camerakit-android` is using `JNI to libjpeg-turbo` (according to @mcorner).

joshuafarray, 2018-12-26:
@n1ru4l Thanks for the additional info.  Will investigate further and report back if I can figure anything useful out.

n1ru4l, 2018-12-27:
@benhartouz As I mentioned this should only be a temporary solution. If you want this issue resolved you should investigate on how to do the image processing without high memory usage.

I personally use the `skipProcessing` option, so it is very unlikely that I am going to fix the issue.

benhartouz, 2018-12-27:
@n1ru4l yeah will investigate further.

kassim, 2018-12-31:
I've been experiencing the same crash reports in Crashlytics, where a OOM is produced from the `rotateBitmap` method. On devices Samsung Galaxy J3, Samsung Galaxy Note 3, Sony Xperia E5

I was able to recreate locally on a cheap Samsung Galaxy J1 device after repeatedly trying to perform the same 'take picture' sequence in my app - problem is, some of my users have been reporting not being able to take pictures at all. 

It seems available memory isn't taken into consideration when performing these operations. Perhaps what we need is a callback from the operation that informs us there is not enough memory, and that we need to change parameters (resolution or the aforementioned `skipProcessing` or something else)
    

Here is my stacktrace

```
    java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$3.done(AsyncTask.java:309)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
        at java.util.concurrent.FutureTask.run(FutureTask.java:242)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
        at java.lang.Thread.run(Thread.java:818)
     Caused by: java.lang.OutOfMemoryError: Failed to allocate a 20155404 byte allocation with 8388608 free bytes and 8MB until OOM
        at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
        at android.graphics.Bitmap.nativeCreate(Native Method)
        at android.graphics.Bitmap.createBitmap(Bitmap.java:975)
        at android.graphics.Bitmap.createBitmap(Bitmap.java:946)
        at android.graphics.Bitmap.createBitmap(Bitmap.java:877)
        at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.rotateBitmap(ResolveTakenPictureAsyncTask.java:162)
        at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:92)
        at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:26)
        at android.os.AsyncTask$2.call(AsyncTask.java:295)
        at java.util.concurrent.FutureTask.run(FutureTask.java:237)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588) 
        at java.lang.Thread.run(Thread.java:818) 
```

I'll use `android:largeHeap="true"` for now - thanks for the recommendation.

Samykills, 2019-01-09:
it works with exif

this.camera.takePictureAsync({ quality: 0.2, exif: false, base64: false })

Nocturne-FFG, 2019-01-10:
@mconiglio Solved my problem by adding "compile'c​​om.android.support:exifinterface:27.+"

mconiglio, 2019-01-15:
@benhartouz Seems that `android:largeHeap="true"` did the trick for now. Thanks for your help!

n1ru4l, 2019-01-22:
Crashlog from #2059 (which is a duplicate of this issue)

```
java.lang.RuntimeException
Bitmap.java:975
An error occurred while executing doInBackground()
Jan 21st, 2019, 16:09:13 UTC

STACKTRACE

java.lang.RuntimeException An error occurred while executing doInBackground() 
    AsyncTask.java:309 android.os.AsyncTask$3.done
    FutureTask.java:354 java.util.concurrent.FutureTask.finishCompletion
    FutureTask.java:223 java.util.concurrent.FutureTask.setException
    FutureTask.java:242 java.util.concurrent.FutureTask.run
    ThreadPoolExecutor.java:1113 java.util.concurrent.ThreadPoolExecutor.runWorker
    ThreadPoolExecutor.java:588 java.util.concurrent.ThreadPoolExecutor$Worker.run
    Thread.java:818 java.lang.Thread.run
,

Caused by: java.lang.OutOfMemoryError Failed to allocate a 63701004 byte allocation with 16777216 free bytes and 52MB until OOM 
    VMRuntime.java:-2 dalvik.system.VMRuntime.newNonMovableArray
    Bitmap.java:-2 android.graphics.Bitmap.nativeCreate
    Bitmap.java:975 android.graphics.Bitmap.createBitmap
    Bitmap.java:946 android.graphics.Bitmap.createBitmap
    Bitmap.java:877 android.graphics.Bitmap.createBitmap
    ResolveTakenPictureAsyncTask.java:167 org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.rotateBitmap
    ResolveTakenPictureAsyncTask.java:97 org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground
    ResolveTakenPictureAsyncTask.java:26 org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground
    AsyncTask.java:295 android.os.AsyncTask$2.call
    FutureTask.java:237 java.util.concurrent.FutureTask.run
    ThreadPoolExecutor.java:1113 java.util.concurrent.ThreadPoolExecutor.runWorker
    ThreadPoolExecutor.java:588 java.util.concurrent.ThreadPoolExecutor$Worker.run
    Thread.java:818 java.lang.Thread.run

THREADS

Thread 1
    ThreadedRenderer.java:-2 android.view.ThreadedRenderer.nSyncAndDrawFrame
    ThreadedRenderer.java:386 android.view.ThreadedRenderer.draw
    ViewRootImpl.java:3144 android.view.ViewRootImpl.draw
    ViewRootImpl.java:2943 android.view.ViewRootImpl.performDraw
    ViewRootImpl.java:2525 android.view.ViewRootImpl.performTraversals
    ViewRootImpl.java:1440 android.view.ViewRootImpl.doTraversal
    ViewRootImpl.java:7421 android.view.ViewRootImpl$TraversalRunnable.run
    Choreographer.java:920 android.view.Choreographer$CallbackRecord.run
    Choreographer.java:695 android.view.Choreographer.doCallbacks
    Choreographer.java:631 android.view.Choreographer.doFrame
    Choreographer.java:906 android.view.Choreographer$FrameDisplayEventReceiver.run
    Handler.java:739 android.os.Handler.handleCallback
    Handler.java:95 android.os.Handler.dispatchMessage
    Looper.java:158 android.os.Looper.loop
    ActivityThread.java:7230 android.app.ActivityThread.main
    Method.java:-2 java.lang.reflect.Method.invoke
    ZygoteInit.java:1230 com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run
    ZygoteInit.java:1120 com.android.internal.os.ZygoteInit.main
Thread 22303
Thread 22304
Thread 22305
    Object.java:-2 java.lang.Object.wait
    Daemons.java:162 java.lang.Daemons$ReferenceQueueDaemon.run
    Thread.java:818 java.lang.Thread.run
Thread 22306
    Object.java:-2 java.lang.Object.wait
    Object.java:423 java.lang.Object.wait
    ReferenceQueue.java:101 java.lang.ref.ReferenceQueue.remove
    ReferenceQueue.java:72 java.lang.ref.ReferenceQueue.remove
    Daemons.java:200 java.lang.Daemons$FinalizerDaemon.run
    Thread.java:818 java.lang.Thread.run
Thread 22307
[... 443 lines cut ...]
    ConnectionPool.java:67 okhttp3.ConnectionPool$1.run
    ThreadPoolExecutor.java:1113 java.util.concurrent.ThreadPoolExecutor.runWorker
    ThreadPoolExecutor.java:588 java.util.concurrent.ThreadPoolExecutor$Worker.run
    Thread.java:818 java.lang.Thread.run
Thread 22489
Error reported from this thread
    AsyncTask.java:309 android.os.AsyncTask$3.done
    FutureTask.java:354 java.util.concurrent.FutureTask.finishCompletion
    FutureTask.java:223 java.util.concurrent.FutureTask.setException
    FutureTask.java:242 java.util.concurrent.FutureTask.run
    ThreadPoolExecutor.java:1113 java.util.concurrent.ThreadPoolExecutor.runWorker
    ThreadPoolExecutor.java:588 java.util.concurrent.ThreadPoolExecutor$Worker.run
    Thread.java:818 java.lang.Thread.run
Thread 22490
    Object.java:-2 java.lang.Object.wait
    Thread.java:1220 java.lang.Thread.parkFor$
    Unsafe.java:299 sun.misc.Unsafe.park
    LockSupport.java:158 java.util.concurrent.locks.LockSupport.park
    AbstractQueuedSynchronizer.java:2013 java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await
    LinkedBlockingQueue.java:410 java.util.concurrent.LinkedBlockingQueue.take
    ThreadPoolExecutor.java:1036 java.util.concurrent.ThreadPoolExecutor.getTask
    ThreadPoolExecutor.java:1098 java.util.concurrent.ThreadPoolExecutor.runWorker
    ThreadPoolExecutor.java:588 java.util.concurrent.ThreadPoolExecutor$Worker.run
    Thread.java:818 java.lang.Thread.run
```

pitchourou, 2019-02-16:
any update on this guys ? using `largeHeap=true` is not an ideal solution

pitchourou, 2019-02-28:
Still crashing with latest release 1.12.0

mconiglio, 2019-02-28:
Yes @pitchourou, I've even seen the app crashing because of this with `largeHeap=true`

pitchourou, 2019-02-28:
So , I should not set `largeHeap=true` ??
I have set it to True because the crash was about `OutOfMemoryError`

pitchourou, 2019-02-28:
The crash is related to image processing ? is there a way to skip processsing but still having image data, at least width and height, before I was using `react-native-image-resizer` which process the image , reduces its size pretty well.

mconiglio, 2019-02-28:
I'm using the `largeHeap=true` workaround and it solved the issue at least for now. I'm just saying that I've seen the app crashing one or two times when taking a picture using that approach.

guhungry, 2019-03-03:
```
org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground
ResolveTakenPictureAsyncTask.java, line 116
java.lang.NullPointerException: Attempt to invoke virtual method 'int android.graphics.Bitmap.getWidth()' on a null object reference
```

Crash at ResolveTakenPictureAsyncTask.java#L116 don't know why mBitmap is null here https://github.com/react-native-community/react-native-camera/blob/401c485db342fe6c7bd8580057c82b5ff412170d/android/src/main/java/org/reactnative/camera/tasks/ResolveTakenPictureAsyncTask.java#L116

mdelclaro, 2019-03-28:
`largeHeap=true` seemed to work here too.

closed 2019-06-07 by stale[bot]

jagadeeshym-dev, 2021-07-09:
We are facing crashes for some users, while taking picture. Mainly for oppo device

react-native-camera: ^3.42.3
OS Version: Android 10

Here is my crash logs

java.lang.RuntimeException · startPreview failed
Camera.java:-2 android.hardware.Camera.startPreview	
Camera1.java:795 com.google.android.cameraview.Camera1$8.onPictureTaken	
Camera.java:1364 android.hardware.Camera$EventHandler.handleMessage	
Handler.java:107 android.os.Handler.dispatchMessage	
Looper.java:238 android.os.Looper.loop	
HandlerThread.java:67 android.os.HandlerThread.run