react-native-camera/react-native-camera #1919: Incorrect value reading 47 digit barcode 'interleaved 2 of 5' (Brazilian type)

douglashptiago, 2018-11-12:
When reading bar code in 'interleaved 2 of 5' format, a code with 43 digits is being returned, 4 less than expected.
Code that I need: 00190000090050201018400016324188122550000015000
Code that is returned to me: 0019122550000150000000000502010180001632418
Any way to set the desired barcode length?

[image]
[image]

n1ru4l, 2018-11-14:
Does this occur on both android and ios devices?

grsabreu, 2018-12-21:
As of this, the reading is indeed correct.
https://stackoverflow.com/questions/22234424/zxing-not-reading-brazilian-barcode-bills-interleaved-2-of-5-correctly-how-ca

sibelius, 2018-12-21:
Welcome to Brazil

grsabreu, 2018-12-21:
🙃

n1ru4l, 2018-12-21:
@grsabreu Does this mean we can close this issue?

closed 2019-06-08 by stale[bot]

fakenickels, 2019-06-08:
For anyone looking this in the future, the barcode reading is correct. You gotta convert the read line to a the the `linha digitável` format
https://github.com/Bruno-Louvem/boleto-barcode-parser

r-guimaraes, 2019-10-03:
Great, thanks @fakenickels . Really helpful reference