react-native-camera/react-native-camera #1921: why make offline bundle cound not preview just loading

chenxin0916, 2018-11-14:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

n1ru4l, 2018-11-14:
I cannot understand what you want. Please provide more information

chenxin0916, 2018-11-14:
thank you！i solve it

closed 2018-11-14 by n1ru4l