react-native-camera/react-native-camera #1922: onBarCodeRead don't callback

YouCii, 2018-11-14:
# Question

**To Do First**
- [ok] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ok] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ok] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
有没有中国的小伙伴, 锤子手机运行时onBarCodeRead不回调, 有没有碰见的?

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

n1ru4l, 2018-11-14:
Could you please ask your question in english?

closed 2018-11-14 by sibelius

YouCii, 2018-11-15:
Sorry, but the problem occurs only on a custom Android phone of China. So...