react-native-camera/react-native-camera #1923: Real-time Object detection / Image Labeling Support

tushgup, 2018-11-14:
# Feature-Request

**Describe the Feature**
Just like the onTextRecognized prop, it'd be great to have a real-time object detection/image labeling prop! 

**Possible Implementations**
From what I understand, onTextRecognized uses Firebase ML Kit on-device text OCR, so I'm guessing the model can be switched to their image labeling module.
See: https://firebase.google.com/docs/ml-kit/label-images

n1ru4l, 2018-11-15:
Hey @tushgup 

thank you for your feature request. It would be awesome if you could implement this feature.

brendanyounger, 2018-11-28:
I am interested in implementing this feature. How do you set up development? I don't see a test project in the repo.

sibelius, 2018-11-28:
check this https://github.com/react-native-community/rncamera-example

IssuehuntBot, 2018-12-03:
@issuehuntfest has funded $20.00 to this issue. [See it on IssueHunt](https://issuehunt.io/repos/33218414/issues/1923)

ahmetozalp, 2018-12-04:
+1 plss.

ahmetozalp, 2018-12-04:
help link: [stackoverflow](https://stackoverflow.com/questions/53621240/react-native-camera-real-time-object-detection-image-labeling-support)

n1ru4l, 2018-12-05:
@ahmetozalp This is a open source project that lives from its contributors. It is unlikely this feature will be implemented from any of current maintainer (unless they have a project that requires it). Therefore I encourage you to implement this feature and contribute to this library if you need this feature!

ahmetozalp, 2018-12-05:
use this library if I know java or swift. Thank You!

jazzystring1, 2018-12-10:
I already successfully implemented this feature in my project. Only that, I didn't use MLKit for image labeling because of lack of extensible.  I used TensorFlow Lite (https://www.tensorflow.org/lite) so that I can train model(to improve accuracy) and load it for image recognition establishing flexibility for the user. Tell me what you think about this.

n1ru4l, 2018-12-10:
@jazzystring1 I am not biased on any particular implementation and if preferable I would like to keep it modular, so that if someone wants to use MLKit instead of TensorFlow Lite he can do so.

Maybe related: https://github.com/react-native-community/react-native-camera/issues/1262

tushgup, 2018-12-10:
@jazzystring1 Hey, that's awesome! What do you exactly mean when you say 'lack of extensibility'? I'd love to have a look at your PR!

jazzystring1, 2018-12-10:
@tushgup What I mean is that since MLKit ondevice image labeling(offline use) only supports up to 400 labels while the Cloud API(for online use) only supports 1000 labels per month if I'm not mistaken. What I want to happen in my project is to recognize more than 2000 labels for offline usage without using Cloud API. You can't add or extend the labels on your own in MLKit for ondevice image labeling unless you pay. Thats why I used tensorflow Lite because I can train a model on my own and load it on my phone.

tushgup, 2018-12-10:
@jazzystring1 I see. Yeah, I created the issue because I noticed the similarities in their coding structure, and I feel like the 400 labels is a good starting point for most use-cases. Looking forward to seeing your implementation! ^_^

jazzystring1, 2018-12-10:
Oops I was totally wrong. MLKit can now let you use your own custom model for Tensorflow Lite which I haven't read in the past few months. (https://firebase.google.com/docs/ml-kit/android/use-custom-models). This is exciting. I'm looking forward into this!

ahmetozalp, 2018-12-10:
I'm following the subject++

ahmetozalp, 2018-12-12:
to add this feature required correction?

https://github.com/react-native-community/react-native-camera/blob/master/ios/TextDetector/TextDetectorManager.m

arseniisemenov, 2018-12-24:
@ppsreejith such a great work  👍 ! Can't wait to use it in iOS !

Udbhav12, 2018-12-24:
@ppsreejith -  Can your code detect barcode present in images ?

arseniisemenov, 2018-12-24:
@Udbhav12 barcode scanner is implemented in react-native-camera (this repo)

Udbhav12, 2018-12-24:
@arseniisemenov - I am not talking about barcode scanning,I am already using RN Camera's barcode scanning. I was talking about barcode detection in a static image. I will be first using RN camera to capture an image and once the image has been captured, i need to detect whether it contains barcode or not

tushgup, 2019-01-07:
Hi! Has there been any progress on this yet?

brendanyounger, 2019-01-07:
> Hi! Has there been any progress on this yet?

I've got a fork (https://github.com/brendanyounger/react-native-camera) that works for barcode scanning on both iOS and Android using MLKit.  It's very rough and I don't have any immediate plans to clean it up and submit a PR, but if you want to see how it's done, it's not too hard to follow.

IssuehuntBot, 2019-04-10:
@deeppicklu has funded $2.00 to this issue.

---
- Submit pull request via [IssueHunt](https://issuehunt.io/repos/33218414/issues/1923) to receive this reward.
- Want to contribute? Chip in to this issue via [IssueHunt](https://issuehunt.io/repos/33218414/issues/1923).
- Checkout the [IssueHunt Issue Explorer](https://issuehunt.io/issues) to see more funded issues.
- Need help from developers? [Add your repository](https://issuehunt.io/r/new) on IssueHunt to raise funds.

IssuehuntBot, 2019-04-10:
@deeppicklu has cancelled funding for this issue.(Cancelled amount: $2.00) [See it on IssueHunt](https://issuehunt.io/repos/33218414/issues/1923)

sejas, 2019-06-14:
Interested ! in this feature. How can I help?
Anybody still interested?

sibelius, 2019-06-14:
feel free to implement on android or iOS side

closed 2019-08-20 by stale[bot]