react-native-camera/react-native-camera #1927: video is not playing after saving to gallery, when used recordAsync()

swaroopa94, 2018-11-19:
# Bug Report
**To Do First**
-I used the latest version of this module

**Related Modules**

**Platforms**

**Versions**

- Android:7
- iOS:10
- react-native-camera:^1.4.1
- react-native: 0.53.0
- react:16.2.0

**Description/Current Behaviour**
I recorded the video using recordAsync() method It returned a uri and I saved i to local gallery but the video is not at all playing showing 0 sec and o bytes

**Expected Behaviour**
Expedted is to play the video that is saved

**Steps to Reproduce**
 I have added this code :
` const { uri, codec = "mp4" } = await this.camera.recordAsync(options);
     const type = `video/${codec}`;
      const data = new FormData();
      data.append("video", {
        name: "mobile-video-upload",
        type,
        uri
      });
      console.log("videouri",uri)`

I got this type of url : file:///data/user/0/com.mla/cache/Camera/75831496-cf6d-4150-816a-f682c25442b8.mp4

for video recording and stopped using stopRecording method

and for saving to gallery using
 `CameraRoll.saveToCameraRoll(uri).then(r => {
         console.log("uriuriuriuriuriuri",r)
          })
         .catch((err) => {
            console.log("errerrerrerr",err)
           });`

returned a url like this:  content://media/external/video/media/7398

But the video is not at all playing in android showing cannot play the video.
In iOS the video is not saving in the gallery

AlexSeoven, 2018-11-19:
@swaroopa94, i have similar problem
but after request permissions i can save video on android
try, maybe it helps

```
async componentDidMount() {
   await this.checkPermission()
};

async checkPermission() {
      if (Platform.OS !== 'android') {
          return Promise.resolve(true);
      }
  
      let result;
      try {
          result = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.RECORD_AUDIO, { title:'Microphone Permission', message:'App needs access to your microphone' });
          result2 = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.WRITE_EXTERNAL_STORAGE, { title:'Storage Permission', message:'App needs access to your storage' });
      } catch(error) {
          console.error('failed getting permission, result:', result);
      }
      return (result === true || result === PermissionsAndroid.RESULTS.GRANTED && result2 === true || result2 === PermissionsAndroid.RESULTS.GRANTED);
    }`

sibelius, 2018-11-19:
we need to improve docs about permissions

this package won't ask for all the permissions, write permission should be asked in user landed, as not will save video in galerry for instance

sibelius, 2018-11-19:
We save the video in a temporary location, so you don't need permission for that

but when you want to move to gallery or process the video you need the permission

this is how Expo Camera was implemented, that's why is kind of tricky

n1ru4l, 2018-11-19:
I will send a PR to the docs with the snippet that I am using!

EthanDM, 2019-01-21:
I'm currently having this issue as well, even after asking for storage permission during runtime.

Was anything ever discovered on this? The videos are saving but are unplayable and 0:00 long.

closed 2019-06-08 by stale[bot]