react-native-camera/react-native-camera #1930: pauseAfterCapture option not working

equesteo, 2018-11-19:
# Bug Report
**To Do First**
- [X] Did you try latest release?
- [ ] Did you try master?
- [X] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
Android

**Versions**

- Android: 7.1.2
- iOS:
- react-native-camera: 1.4.1
- react-native: 0.55.4
- react: 16.2.0

- react-native-navigation: 2.0.2556

**Description/Current Behaviour**
When taking a picture using RNCamera.takePictureAsync, I pass the option `pauseAfterCapture` and expect the viewfinder to pause after the photo is captured. It never pauses, even though the photo is captured.

**Expected Behaviour**
The viewfinder should pause upon capture, then be unpaused when I call `resumePreview`.

**Steps to Reproduce**
My camera looks like this: 
```
        <View style={{width, height: width}}>
          { this.props.showCam ? <RNCamera
            ref={ref => { this.camera = ref }}
            style = {styles.preview}
            type={this.state.backCamera ? RNCamera.Constants.Type.back : RNCamera.Constants.Type.front}
            flashMode={RNCamera.Constants.FlashMode.off}
          /> : null}
        </View>
```

And taking a picture looks like this:
```
takePicture () {
    if (this.camera) {
      this.camera.takePictureAsync({pauseAfterCapture: true}).then((data) => {
        const cropData = {
          offset: {
            x: 0,
            y: 0,
          },
          size: {
            width: data.width,
            height: data.width,
          },
        };
        ImageEditor.cropImage(
          data.uri,
          cropData,
          (successURI) => {
            ImagePicker.cleanSingle(data.uri).catch(e => logError(e))
            this.camera.resumePreview() // not working?
            this.props.stashNewRidePhoto(successURI)
          },
          (error) => {
            console.log(error.message);
          },
        );

      }).catch((e) => { logError(e) })
    }
  }
```

n1ru4l, 2018-11-20:
:tada: This issue has been resolved in version 1.4.3 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.4.3)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

galcivar, 2018-11-21:
I am experiencing this issue using the docs example from the [Docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md). Just changed options:
`const options = { quality: 1, base64: true, pauseAfterCapture: true }`

Packages:
"react-native": "^0.57.5",
"react-native-camera": "^1.4.3"

ilibilibom, 2019-11-20:
This is still happening to me with version ^3.11.0
error: 
```
Error: Preview is paused - resume it before taking a picture.
```
My function: 
```
takePicture = async() => {
    if (this.camera) {
      const options = { quality: 0.5, base64: true, pauseAfterCapture: true };
      const data = await this.camera.takePictureAsync(options);
    }
  };
```

matheuszenker, 2019-12-19:
Mine too, using the same example.

matheuszenker, 2019-12-19:
I'm getting the same error. 

"react-native": "0.61.5",
 "react-native-camera": "^3.15.0",

VictorMion, 2020-01-19:
I'm having the same error.

`"react-native": "0.61.5",
    "react-native-camera": "^3.15.1",`

I'm following the documentation tutorial.

kevintbradbury, 2020-02-05:
i have the opposite issue, even if pauseAfterCapture = false (which is set to false by default) set as an option, camera freezes each time it takes a photo. 

react: 16.9.0
react-native: 0.61.5
react-native-camera: git+https://git@github.com/react-native-community/react-native-camera.git

closed 2020-04-13 by stale[bot]

felansu, 2021-07-08:
its happening in:

"react": "17.0.1"
"react-dom": "17.0.2"
"react-native": "0.64.1"
"react-native-camera": "3.44.0"

PR #1933 (2018-11-20): Fixes pauseAfterCapture for Camera1  #1930
Stops the preview if pauseAfterCapture option is enabled.
Fixes #1930
commit 71b10c1a (2018-11-20): fix(android): stop preview if pauseAfterCapture is set to true (#1933)

Fixes #1930