react-native-camera/react-native-camera #1934: The camera view flickers with a dark blue mask

lucaspang, 2018-11-20:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
iOS

**Versions**
- iOS: 12.0
- react-native-camera: 1.4.2
- react-native: 0.57.2
- react: 16.5.2

**Description/Current Behaviour**
The camera area flickers with a dark blue mask for about 0.1 second whenever I called recordAsync and stopRecording

**Expected Behaviour**
No mask should be covering the camera view

**Steps to Reproduce**
The camera component: 
`<RNCamera
          ref={ref => {
            this.camera = ref
          }}
          style={{
            flex: 1
          }}
          type={RNCamera.Constants.Type.back}
          flashMode={RNCamera.Constants.FlashMode.off}>
          {({ camera, status }) => {
            if (status !== 'READY')
              return <View style={{ flex: 1, backgroundColor: 'blue' }} />
            return (
              <View
                style={{
                  flex: 1,
                  flexDirection: 'row',
                  justifyContent: 'center',
                  alignItems: 'flex-end'
                }}>
                <TouchableOpacity
                  style={{ height: 50 }}
                  onPress={this.buttonClick}>
                  <Text>
                    {this.state.isRecording == true ? 'Stop' : 'Start'}
                  </Text>
                </TouchableOpacity>
              </View>
            )
          }}
        </RNCamera>`

The start recording function:
const { uri, codec = 'mp4' } = await this.camera.recordAsync()
const result = await CameraRoll.saveToCameraRoll(uri, 'video')

The stop recording function:
this.camera.stopRecording()

**Additionals**
Here are some screen cap to help you understand what is happening:
I am filming the following scene
[image]

When I called recordAsync and stopRecording, the view follows with a blue mask or something on the view, sometimes the recorded video captured the blue mask.
[image]

n1ru4l, 2018-11-20:
@listicos is this caused by your pull request? (#1886)

This should definitely not be the default behaviour.

@lucaspang What happens if you use the [`defaultVideoQuality`](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md#ios-defaultvideoquality) property?

maxschmeling, 2018-11-22:
I believe this is being caused by the `updateSessionAudioIsMuted` call in the `record` method of `RNCamera`.

If I comment out this line, the flicker goes away: https://github.com/react-native-community/react-native-camera/blob/b1d91eb91ab148356a2069e20b24c93068154b2f/ios/RN/RNCamera.m#L525

I'm going to attempt to call `updateSessionAudioIsMuted` in `setupMovieFileCapture` and see if that resolves the issue. I'm not sure if there's a reason the audio input isn't setup until `record` is called (via `recordAsync` on the JS side).  Maybe there's a reason it can't be done when `setupMovieFileCapture` is called?

n1ru4l, 2018-11-22:
Okay https://github.com/react-native-community/react-native-camera/pull/1932 is the culprit.

@maxschmeling Yeah Lets introduce the `defaultMuted` property! Do you want to send a pull request?

maxschmeling, 2018-11-22:
I could be wrong, but I don't think #1932 is the culprit. It was still calling `updateSessionAudioMuted` before that point. #1932 just changed the value it was passing.

Again, I could be misunderstanding (not an Obj-C developer at all) but it doesn't matter too much either way. The `defaultMuted` change should fix it.

maxschmeling, 2018-11-22:
Sorry to flood with comments here but don't want to forget things through the holidays. I am seeing the flicker when `stopRecording` is called for the first time even after my POC change for calling `updateSessionAudioIsMuted` earlier. I'm assuming it's removing the audio input after recording stops, but not sure (though maybe that isn't the case because I don't see it flicker again if you start recording a second time on the same component).

I'll dig into that as well as soon as I get a chance and hopefully include a fix for that in the PR.

n1ru4l, 2018-11-22:
@maxschmeling IDK how the `defaultVideoQuality` property behaves today when updated, but it would be nice to have a change of the `defaultVideoQuality` property to be reflected on the native camera instance.

And thank you for your interest in contributing to this project! Ping me in the PR and I will review and test the implementation 😊

maxschmeling, 2018-11-22:
@n1ru4l I've submitted a WIP pull request that has the initial fix in it but a couple todos for the flicker on the first `stopRecording` and looking into handling changes in the `defaultVideoQuality` and `defaultMuted` values. I'll work on those as I have time today and tomorrow.

https://github.com/react-native-community/react-native-camera/pull/1942

erikkjernlie, 2020-08-14:
Still no workaround here?