react-native-camera/react-native-camera #1935: add tests using react-native-testing-library

sibelius, 2018-11-20:
# Feature-Request

https://github.com/callstack/react-native-testing-library

n1ru4l, 2018-11-20:
What exactly should be tested?

sibelius, 2018-11-20:
- [ ] it should render with correct props
- [ ] it should render correctly with render props

maybe more

SimonErm, 2018-11-20:
Maybe we could add a test app and use https://github.com/wix/detox. Furthermore the test app could be used for static analysis like https://sonarcloud.io/about/sq

sibelius, 2018-11-20:
related to https://github.com/react-native-community/react-native-camera/issues/1620

Detox would really test end to end this component

n1ru4l, 2018-11-20:
Can Detox test hardware? Most issues with this project can unfortunately only be tested manual 😢

sibelius, 2018-11-20:
it can run on devices

like this AWS Device Farm

n1ru4l, 2018-11-20:
That is pretty rad but we would need a sponsor first!

teggno, 2019-02-08:
> Can Detox test hardware? Most issues with this project can unfortunately only be tested manually😢

It would be possible to use Microsoft's Azure Pipelines for running detox tests on the **Android emulator** an on **iOS simulator**. Azure Pipelines is Microsoft's CI/CD offering and it's free for open source projects (including parallel build jobs). We've successfully set up such tests in a project where we are using react native.

closed 2019-06-08 by stale[bot]