react-native-camera/react-native-camera #1943: App crash after linking

achimbaggins, 2018-11-23:
# Question

**To Do First**
- [ ok ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [  ok ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [  ok ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
Crash After Build
I was use this package before with RN 0.49.2 and run as expected
now i try to init a new project with RN 56, and app run as expected, but something happen when I yarn add of this package, especially every linking this package... after linking build process successful but when the app launch, the app force close
and I try to unlink the package without remove package and build success and app run well but I can't use this package :( 
I try to log using sentry, there is no report there... somebody can help me?

**Tags**
crashafterlinking
> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

SimonErm, 2018-11-23:
If this just happens on Android, it sounds like missing permission at manifest or missing permission request at runtime.

n1ru4l, 2018-12-12:
Closed due to missing feedback.

closed 2018-12-12 by n1ru4l