react-native-camera/react-native-camera #1946: Manual Focus Depth not working?

kosintop, 2018-11-24:
**Related Modules**

**Platforms**

**Versions**
- react-native-camera: 1.4.3
- react-native: 0.5
- react:16.0.0

**Description/Current Behaviour**
The camera preview does not change focus after adjusting focus level, all focusDepth look blur the same

**Expected Behaviour**
The camera preview focus/blur level should change when focusDepth change

**Steps to Reproduce**

set <RNCamera> autoFocus to RNCamera.Constants.AutoFocus.off
adjust focusDepth between 0-1

kosintop, 2018-12-01:
After googling around it looks like the device's camera has to support manual focus
I install this [apk ](https://play.google.com/store/apps/details?id=com.sdgsystems.camera2probe&rdid=com.sdgsystems.camera2probe&pli=1)on my phone to check my camera capabilities and turn out it does not support manual focus

closed 2018-12-01 by kosintop

n1ru4l, 2018-12-02:
Feel free to send a pull request that adds this information to the docs!