react-native-camera/react-native-camera #1947: RNSensorOrientationChecker undefined upgrading 1.2.0 to 1.4.3

jtwarren, 2018-11-25:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**

RNCamera

**Platforms**

iOS

**Versions**

- iOS: 12.1
- react-native-camera: 1.4.3
- react-native: 0.57.5
- react: 16.6.1

**Description/Current Behaviour**

After upgrading from 1.2.0 to 1.4.3 I am getting a build error on ios.
```bash
Undefined symbols for architecture arm64:
  "_OBJC_CLASS_$_RNSensorOrientationChecker", referenced from:

      objc-class-ref in libRNCamera.a(RNCamera.o)

ld: symbol(s) not found for architecture arm64

clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

This is resolved by going back down to 1.2.0. I was sure to unlink, uninstall, reinstall, relink, and even try to manually link the new version. Anything else I'm missing?

SychevSP, 2018-11-28:
I had the same issue after upgrading from 1.2.0 to 1.4.3.
I solved it by deleting FaceDetector folder in Project Navigator in xcode.
I found the solution in the manual:
https://github.com/react-native-community/react-native-camera#face-detection-or-text-recognition-steps
Hope this helps

sfeidersullivan, 2018-11-29:
Also getting (almost) the same error after upgrading from 1.1.4 -> 1.4.3:
```
Undefined symbols for architecture x86_64:
  "_OBJC_CLASS_$_RNSensorOrientationChecker", referenced from:

      objc-class-ref in libRNCamera.a(RNCamera.o)

ld: symbol(s) not found for architecture x86_64

clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

- iOS: 12.1
- react-native-camera: 1.4.3
- react-native: 0.56.0

While @SychevSP answer does work, why is this version of RN Camera purposefully breaking out of the box?
Where you either have to add the GMV frameworks or manually opt out of Face Detection with a postinstall script.

closed 2019-06-08 by stale[bot]