react-native-camera/react-native-camera #1948: [ios] App crashes, permission changes made to Info.plist file automatically removed on build.

hautch, 2018-11-25:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**

**Versions**
- Android:
- iOS: 12.0
- react-native-camera: ^1.4.3 
- react-native: 0.57.3
- react: 16.6.0-alpha.8af6728

**Description/Current Behaviour**
When my screen with the RNCamera is to be rendered, the app crashes and exits. I have added the necessary permissions to my Info.plist, so it looks like this:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>LastAccessedDate</key>
	<date>2018-11-25T20:15:24Z</date>
	<key>WorkspacePath</key>
	<string>/Users/rlveiga/Work/trampolin-front/ios/Trampolin.xcodeproj</string>

	<key>NSCameraUsageDescription</key>
	<string>Your message to user when the camera is accessed for the first time</string>
</dict>
</plist>
```

But when running `react-native run-ios`, the changes made to Info.plist are gone, and the app crashes. What's going on?

hautch, 2018-11-26:
I have fixed this by including Privacy - Camera Usage Description to the Custom iOS Target Properties in Xcode. Why isn't this in the docs?

n1ru4l, 2018-12-12:
@hautch Feel free to send a pull request that improves the documentation!

closed 2019-06-08 by stale[bot]