react-native-camera/react-native-camera #1949: pauseAfterCapture not fully functional on 1.4.3

ntelkedzhiev, 2018-11-26:
**Platforms**

**Versions**
- Android: 7.1.2
- react-native-camera: 1.4.3
- react-native: 0.57.5
- react: 16.6.3

**Description/Current Behaviour**
There is a 2-3 seconds delay before the image "freezes".

**Expected Behaviour**
Image should "freeze" on capture.

**Steps to Reproduce**
Set `pauseAfterCapture: true`

ivosabev, 2019-02-12:
Any progress on this?

n1ru4l, 2019-02-13:
@ivosabev You could start by using git bisect in order to find out when this issue was introduced (https://git-scm.com/docs/git-bisect). It would help fixing this issue a lot.

Otherwise once somebody is working on this he will probably notify in this thread. No need to ask for progress.

ntelkedzhiev, 2019-02-13:
@n1ru4l This issue has existed ever since `pauseAfterCapture` was introduced (https://github.com/react-native-community/react-native-camera/commit/f6f9f29). https://github.com/react-native-community/react-native-camera/commit/71b10c1 made it better but there is still a significant delay.

closed 2019-06-08 by stale[bot]