react-native-camera/react-native-camera #1950: iOS: [UIApplication statusBarOrientation] must be used from main thread only

n1ru4l, 2018-11-26:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
iOS

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:

**Description/Current Behaviour**

Runtime Warning:

```- (UIInterfaceOrientation)getOrientationBy:(CMAcceleration)acceleration``` in `RNSensorOrientationChecker.m`

-[UIApplication statusBarOrientation] must be used from main thread only

**Expected Behaviour**
No runtime warning

**Steps to Reproduce**
Use rn-camera.

**Additionals**

[image]

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

justinkiang, 2018-11-29:
I believe this issue is causing barcode recognition to not work when I turn the device landscape in an app that is locked to portrait

n1ru4l, 2018-11-30:
@justinkiang Wanna contribute a fix?

justinkiang, 2018-11-30:
That should do it

closed 2018-12-19 by n1ru4l