react-native-camera/react-native-camera #1951: What is the proper way to import for flowtype to work?

maxschmeling, 2018-11-26:
# Question

In trying to get flow typing in my project for react-native-camera, it seems I have to import like this:
```
import RNCamera from "react-native-camera/src/RNCamera";
```
instead of what all the docs say which is:
```
import { RNCamera } from "react-native-camera";
```
Am I missing something? Is flow not configured correctly? I notice that `src/index.js` doesn't have `// @flow` at the top and if I add it locally, everything works with the standard import.

Is there a reason the `index.js` file doesn't have `// @flow`?

**Tags**
static-typing, flow-type, flow

maxschmeling, 2018-11-26:
I just realized that the import change causes the packager build to fail, so it doesn't actually work even though the flow typing works great that way.

n1ru4l, 2018-12-19:
@sibelius Can you answer this question?

sibelius, 2018-12-19:
this should work:

```jsx
import { RNCamera } from "react-native-camera";
```

feel free to send a pr to fix or improve this

I recommend moving to TS

closed 2018-12-19 by sibelius