react-native-camera/react-native-camera #1954: start failed - recordAsync crashes

dasouch, 2018-11-28:
# Bug Report
**To Do First**
- [ *] Did you try latest release?
- [ *] Did you try master?

**Related Modules**

RNCamera

**Platforms**
Android

**Versions**

- Android: 7.1.1
- react-native-camera: 1.4.3
- react-native: 0.57.4
- react: 16.6.0

**Description/Current Behaviour**
When starting to record, Error appears: start failed.

[image]

**Steps to Reproduce**
```
takeVideo = async() => {
    this.state.animated.setValue(0)
    Animated.timing(this.state.animated, {
      toValue: 1,
      duration: 17000,
    }).start()
    const data = await this.camera.recordAsync({
        quality: RNCamera.Constants.VideoQuality['480p'],
        mirrorVideo: true,
        maxDuration: 15
      })
    const content = { data: data.uri, type: 'video' }
  }
```
```
<RNCamera
    ref={ref => {this.camera = ref}}
    style={[styles.camera.camera]}
    type={RNCamera.Constants.Type.back}
    flashMode={false}
    captureAudio
    permissionDialogTitle={'Permiso para usar la cámara'}
    permissionDialogMessage={'Necesitamos su permiso para usar su teléfono con cámara'}
    autoFocus={RNCamera.Constants.AutoFocus.on}
/>
```

ErikaMolino, 2018-12-06:
Hi, i have the same issue.
You solved it?

ErikaMolino, 2018-12-06:
@dasouch thank you!!!!!! you made my day.

n1ru4l, 2018-12-18:
Use a `VideoQuality` that is available on your device.