react-native-camera/react-native-camera #1956: Front camera shows black screen and crashes app on iOS

albullington, 2018-11-28:
# Bug Report
**To Do First**
- [ Y ] Did you try latest release?
- [ Y ] Did you try master?
- [ Y ] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
iOS

**Versions**
- Android: N/A
- iOS: 12, testing production build on an iPhone SE
- react-native-camera: git+https://github.com/react-native-community/react-native-camera.git#e18d52d2045f7db82c11368ff51cc3882d002e75
- react-native: 0.57.1
- react: 16.5.0
- react-navigation: ^2.17.0

**Description/Current Behaviour**
1. When toggling to the front RNCamera from the back RNCamera, the screen appears black. If I press the button to take photos, the app crashes. 
2. If I set the initial RNCamera type prop to "front," the app immediately crashes when rendering the camera screen. 

**Expected Behaviour**
User should be able to toggle between the front and back camera. When toggling to the front camera, users should be able to see the camera, not a black screen. Users should be able to take a photo while the front camera is toggled on. 

**Steps to Reproduce**
1. Create a production build on iOS using "react-native run-ios --configuration Release."
2. Toggle from the back camera to the front camera OR set RNCamera prop to "front."

**Additionals**
[image]
This is what displayed after I toggled to the front camera via the upper right button. 

```
const flashModeOrder = {
    off: "on",
    on: "off"
};

class CameraScreen extends Component {
  constructor( { navigation }: Props ) {
    super();

    this.state = {
      cameraType: "back",
      flash: "off",
      flashText: "OFF",
    };

    this.toggleCamera = this.toggleCamera.bind( this );
    this.toggleFlash = this.toggleFlash.bind( this );
  }

  takePicture = async () => {
    if ( this.camera ) {
      this.camera.takePictureAsync()
        .then( data => this.savePhotoToGallery( data ) )
        .catch( ( err ) => {
          this.setState( {
            error: err.message
          } );
        } );
    }
  }

  toggleFlash() {
    const { flash } = this.state;

    this.setState( {
      flash: flashModeOrder[flash],
      flashText: flashModeOrder[flash].toUpperCase()
    } );
  }

  toggleCamera() {
    const { cameraType } = this.state;

    this.setState( {
      cameraType: cameraType === "back" ? "front" : "back"
    } );
  }

  render() {
    const {
      cameraType,
      flash,
      flashText,
      loading
    } = this.state;

    const { navigation } = this.props;

    return (
      <RNCamera
        ref={( ref ) => {
          this.camera = ref;
        }}
        type={cameraType}
        style={{ flex: 1 }}
        flashMode={flash}
        permissionDialogTitle="Permission to use camera"
        permissionDialogMessage="We need your permission to use your camera phone"
      >
        <CameraTopNav
          navigation={navigation}
          cameraType={cameraType}
          flashText={flashText}
          toggleFlash={this.toggleFlash}
          toggleCamera={this.toggleCamera}
        />
        {loading ? <LoadingWheel /> : (
          <View>
            <View />
            <View>
              <TouchableOpacity
                onPress={() => this.takePicture()}
              />
            </View>
          </View>
        )}
      </RNCamera>
    );
  }
}

export default CameraScreen;

const CameraTopNav = ( {
    flashText,
    navigation,
    toggleFlash,
    toggleCamera
  }: Props ) => (
  <View>
    <View>
      <TouchableOpacity
        onPress={() => navigation.push( "Main" )}
      >
        <Text>{navigation.state.key === "PHOTOS" ? exitIconGray : exitIconWhite}</Text>
      </TouchableOpacity>
      <TouchableOpacity
        onPress={() => toggleFlash()}
      >
        {navigation.state.key === "CAMERA" ? (
          <Text>
            {flashText === "OFF" ? flashOffIcon : flashOnIcon}
            {flashText}
          </Text>
        ) : null}
      </TouchableOpacity>
      <TouchableOpacity
        onPress={() => toggleCamera()}
      >
        {navigation.state.key === "CAMERA" ? (
          <Text>{cameraFlipIcon}</Text>
        ) : null}
      </TouchableOpacity>
    </View>
  </View>
);

export default CameraTopNav;
```

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

gaurav321garg, 2019-01-05:
> Could you please provide a crash log?
 I am also facing similar issue.
`Application threw exception NSInternalInconsistencyException: *** -[AVCaptureStillImageOutput captureStillImageAsynchronouslyFromConnection:completionHandler:] Inconsistent state
Originated at or in a subcall of realm::Column<double>::average(unsigned long, unsigned long, unsigned long, unsigned long*) const`

andcosta, 2019-01-12:
it's working now : )

<key>NSMicrophoneUsageDescription</key>
<string>Your message to user when the microphone is accessed for the first time</string>

KryDos, 2019-01-15:
hey @andcosta ,
what did you do to make it working? Have you added NSMicrophoneUsageDescription key to your Info.plist? It didn't help in my case (it was there).

KryDos, 2019-01-16:
Based on what I've found (and based on @gaurav321garg's comment) the crash happens because there is uncaught exception thrown in RN/RNCamera.m. Specifically `takePicture` method. 

At the time when code execution reach `self.stillImageOutput captureStillImageAsynchronouslyFromConnection:connection completionHandler...` part of the code, the exception is happening. There is no `completionHandler` execution because of exception and error handling inside `completionHandler` isn't working, obviously. 

In my case, simple try/catch around the code part mentioned above did the trick. I just call `reject` function inside the catch block and javascript part is handling the error. No crash. Whew. 

Due to my poor knowledge of Objective C it's pretty hard for me to find the actual reason of exception. I'm not sure if `try/catch` solution is actually something that can be accepted as PR. Let me know and I'll send it if needed.

closed 2019-06-08 by stale[bot]