react-native-camera/react-native-camera #1960: Android crashes while taking a photo

Nocturne-FFG, 2018-12-01:
# Bug Report
**To Do First**
- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Related Modules**

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:1.4.3
- react-native:0.52.2
- react:16.2

**Description/Current Behaviour**
Android crashes while taking a photo

**Expected Behaviour**

normal operation

**Steps to Reproduce**
Use the version of react native 0.52.2 and the version of react-native-camera 1.4.3, then take a photo on Android

**Additionals**
After taking the photo, the program crashes directly and quits. So far I have not succeeded. Below is my code.
[image]

[image]

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

Nocturne-FFG, 2018-12-01:
manifest.xml:
[image]

n1ru4l, 2018-12-02:
Could you please provide some logs? You can use `adb logcat` (https://developer.android.com/studio/command-line/logcat)

Nocturne-FFG, 2018-12-03:
@n1ru4l Thank you, I found the answer, add "compile 'com.android.support:exifinterface:27.+'"

closed 2018-12-03 by Nocturne-FFG